feelfreelinux/cspot #2: Zeroconf auth support + auth token persistence

feelfreelinux, 2020-09-28:
Add support to authentication via zeroconf/mdns. 
- Launch a server on random port handling GET `getInfo` and `addUser`
- Solve challenge based on addUser data
- Resulting token can then be used with Session

we should also persist auth token blobs in targets instead of relogging every single time

feelfreelinux, 2020-10-09:
Working implementation in https://github.com/feelfreelinux/cspot/commit/66910743c46f19a440819bc62903c11b26dc75cd

feelfreelinux, 2020-10-09:
Added MbedTLS implementation in: https://github.com/feelfreelinux/cspot/commit/20a0f6a8478db1ba3a8058c55742b1533f8302fd

feelfreelinux, 2020-10-09:
Will merge once confirmed working on esp-idf platform

feelfreelinux, 2020-10-09:
Closed in https://github.com/feelfreelinux/cspot/pull/26

closed 2020-10-09 by feelfreelinux

alufers, 2020-10-09:
[image]