feelfreelinux/cspot #45: Timeouts during authentication

danieltwagner, 2021-03-15:
Hi,

First, thank you very much for creating this project, it does exactly what I'm looking for!
Unfortunately my cli target seems to be hitting repeated timeouts during authentication, both on MacOS as well as inside the Docker container where I tried to run it. Only once did authentication succeed and in all my other attempts it gets stuck with what looks to me like timeouts on the socket's `recv()` call.

```
Connecting with AP <guc3-accesspoint-e-dxpq.ap.spotify.com:4070>
Connected to spotify server
Hello hi
[...]
yes packet
recv
dtw: in PlainConnection::recvPacket(), about to read 4 blocks...
dtw: case EAGAIN, timeoutHandler is not set
dtw: case EAGAIN, timeoutHandler is not set
[...]
```

I imagine you did not encounter this issue during your use, as the code in its current form would hit
```
terminating with uncaught exception of type std::__1::bad_function_call: std::exception
```
with associated backtrace
```
Catchpoint 1 (exception thrown), 0x00007fbaab4bd762 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007fbaab4bd762 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007fbaab4b4590 in std::__throw_bad_function_call() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00005633aeec3609 in std::function<bool ()>::operator()() const (this=0x5633b016f398) at /usr/include/c++/9/bits/std_function.h:687
#3  0x00005633aeec31a4 in PlainConnection::readBlock (this=0x5633b016f390, size=4) at /src/cspot/cspot/src/PlainConnection.cpp:113
#4  0x00005633aeec2e5e in PlainConnection::recvPacket (this=0x5633b016f390) at /src/cspot/cspot/src/PlainConnection.cpp:71
#5  0x00005633aee7a4e3 in Session::processAPHelloResponse (this=0x5633b018d980, helloPacket=std::vector of length 147, capacity 147 = {...}) at /src/cspot/cspot/src/Session.cpp:80
#6  0x00005633aee79f8f in Session::connectWithRandomAp (this=0x5633b018d980) at /src/cspot/cspot/src/Session.cpp:30
#7  0x00005633aee56829 in main (argc=5, argv=0x7ffd0a50d418) at /src/cspot/targets/cli/main.cpp:76
```
This happens as my auth timeouts occur before a `MercuryManager` is instantiated and manages the connection. During instantiation, `MercuryManager` attaches the `timeoutHandler` to the underlying connection, and before that we would attempt to invoke `timeoutHandler()`, a null function pointer. The way I chose to apply a bandaid in my code is
```
diff --git a/cspot/src/PlainConnection.cpp b/cspot/src/PlainConnection.cpp
index b00e6fd..b620599 100644
--- a/cspot/src/PlainConnection.cpp
+++ b/cspot/src/PlainConnection.cpp
@@ -110,7 +110,7 @@ std::vector<uint8_t> PlainConnection::readBlock(size_t size)
             {
             case EAGAIN:
             case ETIMEDOUT:
-                if (timeoutHandler())
+                if (timeoutHandler && timeoutHandler())
                 {
                     printf("Throwing cuz reconnection\n");
                     throw std::runtime_error("Reconnection required");
```

The above patch implies that we'll get stuck forever retrying instead of a potentially better fix where we eventually throw, but I don't see how that would cause the above issue.

I'm not at all familiar with SpotifyConnect and would be very grateful for any pointers or help you could provide to resolve this.

Thanks!

feelfreelinux, 2021-03-15:
Hi, thanks for reporting this. I'll look further into it asap. 
I actually remember this issue happening on Linux machines, somehow caused by ALSASink (probably blocking up a thread and preventing proper auth) but I'm not sure if its related

danieltwagner, 2021-03-15:
Thank you! Please do let me know if I can provide any additional information. I'll also try and make some forward progress but I'm a little bit out of ideas. It happens equally on MacOS and Linux and also when I just enable the `NamedPipeAudioSink` (which makes sense now that I think of it, as that's only initialized after auth completes)

danieltwagner, 2021-03-20:
Some progress. The protobuf serialization doesn't seem to work correctly. The uin32 field `clientHello.login_crypto_hello.diffie_hellman->server_keys_known` does not get serialized. My (failing) messages have a length of 147 bytes, and a single successful message that I captured had length 150 bytes, which included that particular field.

Looking some more, https://github.com/feelfreelinux/cspot/blob/5f0b9298fceb716b49da44d655b6eea6068f66a7/cspot/src/Protobuf.cpp#L174 does not seem to fire and instead it falls out of the series of conditionals (you can test by adding an `else` at the end). I can bandaid it by expanding the conditional to check if `info->typeID` is the same as the `Uint32` type id, in my case `17`. When doing so the field gets serialized correctly and authorization proceeds.

I haven't yet dug further why `any.is<uint32_t>()` doesn't work, nor do I know why it worked at some point on a random try, but this seems to be where the issue is. I might spend some more time digging further tomorrow.

feelfreelinux, 2021-03-20:
> Some progress. The protobuf serialization doesn't seem to work correctly. The uin32 field `clientHello.login_crypto_hello.diffie_hellman->server_keys_known` does not get serialized. My (failing) [...]

Good job on debugging the issue. We've introduced our own reflection system quite recently, so there's a good chance that it's a bit wonky. Could you try to checkout this commit: https://github.com/feelfreelinux/cspot/commit/0175013449fb7e8a809bd2ef937ceb07a79e360c and see if it works for you? It uses nanopb, a completely different protobuf implementation.

danieltwagner, 2021-03-20:
Thanks for the pointer. 0175013449fb7e8a809bd2ef937ceb07a79e360c works for me without modifications, yes.

feelfreelinux, 2021-03-20:
> Thanks for the pointer. [0175013](https://github.com/feelfreelinux/cspot/commit/0175013449fb7e8a809bd2ef937ceb07a79e360c) works for me without modifications, yes.

Great, I guess we pinpointed the issue :) I can't recreate it myself, however, could you provide me with hexdump of data before serialization?
Also feel free to message me on telegram / discord / anywhere, I'm up to work on this issue today

feelfreelinux, 2021-03-30:
Turned out to be an issue with our custom protobuf / reflection generator. Fixed in https://github.com/feelfreelinux/cspot/commit/a3a070117b8629ae670c21c0964938b2a3f43bad

closed 2021-03-30 by feelfreelinux