feelfreelinux/cspot #46: Build failed on Windows (missing 'protos/AnyRefImpl.cpp')

adamhun94, 2021-03-23:
Hi,
I just received my ESP32-A1S audio board for this project, and I'm trying to build a binary for it on Windows.
I installed the SDK and set up VS Code as well. When I'm trying to the build the project (from targets/esp32 folder) I get the following error:
```
CMake Error at ../../cspot/CMakeLists.txt:40 (add_library):
  Cannot find source file:

    protos/AnyRefImpl.cpp
```

I pulled the submodules and installed the python libs (pip3 install protobuf grpcio-tools)
What am I doing wrong?

Thanks!

adamhun94, 2021-03-24:
I checked out https://github.com/feelfreelinux/cspot/commit/0613aaadcaa9ef3da9cfb3f8258e66db5de583de since it's marked as a success build (sorry for missing that).

I was able to build it and flash it, so now I'm testing it. Thanks for your hard work!

feelfreelinux, 2021-03-24:
Hi, thanks for reporting this. We switched to a custom protobuf implementation a while ago so that's probably why the process might be quite wonky. I'll look into it soon :)

philippe44, 2021-03-28:
I was trying to give another try to integrate into squeezelite-esp32 but I had the same issue. I'll wait till you have the time to figure out what's wrong with Windows build as this is my main system

feelfreelinux, 2021-04-07:
> I checked out [0613aaa](https://github.com/feelfreelinux/cspot/commit/0613aaadcaa9ef3da9cfb3f8258e66db5de583de) since it's marked as a success build (sorry for missing that).
> 
> I was able to build it and flash it, so now I'm testing it. Thanks for your hard work!

Actually, docs don't state that but you need to have golang installed and in your path for the new version to work. Let me know if you encounter any other issues on Windows

adamhun94, 2021-04-07:
I just figured it out (and feel a little bit dumb), but I missed **generate_protos.sh** which generates the missing files.
I wrote this batch script for Windows if you don't have git bash:
```
cd ../cpp-reflection
go build -o protoc-gen-cpprefl .
cd ..
mkdir protos
protoc --plugin=protoc-gen-cpprefl=cpp-reflection/protoc-gen-cpprefl --cpprefl_out protos/ protocol/*.proto --proto_path protocol/
```
Place it inside the cmake folder.

I also had to download the [protobuf binary](https://github.com/protocolbuffers/protobuf/releases/latest) and add to PATH

Now I have a successful build from the latest commit.
_(haven't tried it flashing though, I'm not currently at home)_

FYI @philippe44

feelfreelinux, 2021-04-07:
> I just figured it out (and feel a little bit dumb), but I missed **generate_protos.sh** which generates the missing files.
> I wrote this batch script for Windows if you don't have git bash:
>
> ```
> cd [...]

`generate_protos.sh` is an extremely hacky version of doing it and I really need to fix it up.  Right not it doesn't even log the errors properly.

feelfreelinux, 2021-12-22:
Protobuf decoding was reworked in newest master, now works through cmake. Thanks a lot @kuba2k2 :)

closed 2021-12-22 by feelfreelinux