feelfreelinux/cspot #50: Please provide examples of how to set login info in main.cpp comments

geoffreyheller, 2021-04-09:
As per the README file, we need to set the username and password in main.cpp.  Can you please provide an example of how to do this in the main.cpp comments?

Thank you!
-G.

feelfreelinux, 2021-04-09:
Hey, where do the docs state that you need to put in login and password manually? By default running cspot (and always on esp32) will enable zeroconf authenticator, which only requires selecting cspot from the list of available spotify speakers in spotify connect tab. On desktop (using cli) you can also manually authenticate by passing -u username -p password parameters to `cspotcli`.

geoffreyheller, 2021-04-09:
I was looking at the README.md file here:
https://github.com/feelfreelinux/cspot/tree/master/targets/esp32

It seems the README.me is just out of date?

feelfreelinux, 2021-04-09:
You're absolutely right, thanks for pointing it out :)

Resolved in: https://github.com/feelfreelinux/cspot/commit/c414cd6998c5eb6e392ca546da3f1ba3b854c3df

closed 2021-04-09 by feelfreelinux