feelfreelinux/cspot #52: ESP32-WROOM-32 PCM

hajdbo, 2021-04-21:
I use a ESP32-WROOM-32 board with a PCM5102.
The devkit does not have SPRAM.
I replaced the audioSink as:
```
auto audioSink = std::make_shared<PCM5102AudioSink>();
```

CSpot shows up in Spotify connect, but the ESP32 reboots immediately when I select CSpot for play.

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:7476
load:0x40078000,len:13904
load:0x40080400,len:5296
0x40080400: _init at ??:?

entry 0x40080710
I (29) boot: ESP-IDF v4.2-dirty 2nd stage bootloader
I (29) boot: compile time 15:13:37
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) qio_mode: Enabling default flash chip QIO
I (45) boot.esp32: SPI Speed      : 80MHz
I (49) boot.esp32: SPI Mode       : QIO
I (54) boot.esp32: SPI Flash Size : 4MB
I (58) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (82) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  2 factory          factory app      00 00 00010000 00300000
I (97) boot: End of partition table
I (101) boot_comm: chip revision: 1, min. application chip revision: 0
I (108) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3d634 (251444) map
I (191) esp_image: segment 1: paddr=0x0004d65c vaddr=0x3ffb0000 size=0x029bc ( 10684) load
I (195) esp_image: segment 2: paddr=0x00050020 vaddr=0x400d0020 size=0xc5100 (807168) map
0x400d0020: _stext at ??:?

I (436) esp_image: segment 3: paddr=0x00115128 vaddr=0x3ffb29bc size=0x00ef0 (  3824) load
I (437) esp_image: segment 4: paddr=0x00116020 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/Users/Boris/Desktop/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (443) esp_image: segment 5: paddr=0x0011642c vaddr=0x40080404 size=0x10c14 ( 68628) load
I (485) boot: Loaded app from partition at offset 0x10000
I (485) boot: Disabling RNG early entropy source...
I (486) cpu_start: Pro cpu up.
I (489) cpu_start: Application information:
I (494) cpu_start: Project name:     cspot-esp32
I (499) cpu_start: App version:      c414cd6
I (504) cpu_start: Compile time:     Apr 21 2021 15:13:33
I (510) cpu_start: ELF file SHA256:  c99e69cc88a9f0b3...
I (516) cpu_start: ESP-IDF:          v4.2-dirty
I (522) cpu_start: Starting app cpu, entry point is 0x400818bc
0x400818bc: call_start_cpu1 at C:/Users/Boris/Desktop/esp-idf/components/esp32/cpu_start.c:287

I (514) cpu_start: App cpu up.
I (532) heap_init: Initializing. RAM available for dynamic allocation:
I (539) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (545) heap_init: At 3FFBC528 len 00023AD8 (142 KiB): DRAM
I (551) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (558) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (564) heap_init: At 40091018 len 0000EFE8 (59 KiB): IRAM
I (570) cpu_start: Pro cpu start user code
I (597) spi_flash: detected chip: generic
[... 113 lines cut ...]
0x400db73d: ShannonConnection::sendPacket(unsigned char, std::vector<unsigned char, std::allocator<unsigned char> >&) at C:/Users/Boris/Desktop/cspot/cspot/src/ShannonConnection.cpp:36

0x400d8e5a: MercuryManager::execute(MercuryType, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::function<void (std::unique_ptr<MercuryResponse, std::default_delete<MercuryResponse> >)>&, std::function<void (std::unique_ptr<MercuryResponse, std::default_delete<MercuryResponse> >)>&, std::vector<std::vector<unsigned char, std::allocator<unsigned char> >, std::allocator<std::vector<unsigned char, std::allocator<unsigned char> > > >&) at C:/Users/Boris/Desktop/cspot/cspot/src/MercuryManager.cpp:315

0x400d8f41: MercuryManager::execute(MercuryType, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::function<void (std::unique_ptr<MercuryResponse, std::default_delete<MercuryResponse> >)>&, std::vector<std::vector<unsigned char, std::allocator<unsigned char> >, std::allocator<std::vector<unsigned char, std::allocator<unsigned char> > > >&) at C:/Users/Boris/Desktop/cspot/cspot/src/MercuryManager.cpp:323 (discriminator 1)

0x400dc20d: SpircController::sendCmd(MessageType) at C:/Users/Boris/Desktop/cspot/cspot/src/SpircController.cpp:161

0x400dc271: SpircController::notify() at C:/Users/Boris/Desktop/cspot/cspot/src/SpircController.cpp:151

0x400dc40b: SpircController::handleFrame(std::vector<unsigned char, std::allocator<unsigned char> >&) at C:/Users/Boris/Desktop/cspot/cspot/src/SpircController.cpp:129

0x400dc58f: operator() at C:/Users/Boris/Desktop/cspot/cspot/src/SpircController.cpp:30
 (inlined by) _M_invoke at c:\users\boris\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297

0x400d700b: std::function<void (std::unique_ptr<MercuryResponse, std::default_delete<MercuryResponse> >)>::operator()(std::unique_ptr<MercuryResponse, std::default_delete<MercuryResponse> >) const at c:\users\boris\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687

0x400d8579: MercuryManager::handleQueue() at C:/Users/Boris/Desktop/cspot/cspot/src/MercuryManager.cpp:249

0x400d6190: cspotTask(void*) at c:\users\boris\desktop\cspot\targets\esp32\build/../main/main.cpp:59

ELF file SHA256: c99e69cc88a9f0b3

Rebooting...
```

geoffreyheller, 2021-04-21:
I am having a similar issue.   cspot is visible on Spotify but when I go to try to play I get an abort().

In my case I am using an ESP32-WROVER-IB, which is mounted on @fgervais's breakout board: https://github.com/fgervais/esp32-wrover-breakout

I am also using a PCM5102, so I am wondering if that might somehow be a factor.  I implemented my the audiosink declaration the same way as Boris did.

My log looks a bit different than Boris's.  I'm assuming he has additional logging turned on somewhere:

```
Brown�ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7052
ho 0 tail 12 room 4
load:0x40078000,len:15404
load:0x40080400,len:4452
entry 0x40080694
I (29) boot: ESP-IDF v4.4-dev-744-g1cb31e509 2nd stage bootloader
I (29) boot: compile time 20:44:45
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) qio_mode: Enabling default flash chip QIO
I (45) boot.esp32: SPI Speed      : 80MHz
I (50) boot.esp32: SPI Mode       : QIO
I (54) boot.esp32: SPI Flash Size : 4MB
I (59) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  2 factory          factory app      00 00 00010000 00300000
I (98) boot: End of partition table
I (102) boot_comm: chip revision: 1, min. application chip revision: 0
I (109) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=3f430h (259120) map
I (195) esp_image: segment 1: paddr=0004f458 vaddr=3ffbdb60 size=00bc0h (  3008) load
I (197) esp_image: segment 2: paddr=00050020 vaddr=400d0020 size=d30d8h (864472) map
I (460) esp_image: segment 3: paddr=00123100 vaddr=3ffbe720 size=04260h ( 16992) load
I (466) esp_image: segment 4: paddr=00127368 vaddr=40080000 size=1838ch ( 99212) load
I (500) esp_image: segment 5: paddr=0013f6fc vaddr=50000000 size=00010h (    16) load
I (514) boot: Loaded app from partition at offset 0x10000
I (514) boot: Disabling RNG early entropy source...
I (526) psram: This chip is ESP32-D0WD
I (526) spiram: Found 64MBit SPI RAM device
I (526) spiram: SPI RAM mode: flash 80m sram 80m
I (531) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (539) cpu_start: Pro cpu up.
I (542) cpu_start: Starting app cpu, entry point is 0x40081710
I (0) cpu_start: App cpu up.
I (1045) spiram: SPI SRAM memory test OK
I (1053) cpu_start: Pro cpu start user code
I (1054) cpu_start: cpu freq: 240000000
I (1054) cpu_start: Application information:
I (1057) cpu_start: Project name:     cspot-esp32
I (1062) cpu_start: App version:      c414cd6-dirty
I (1068) cpu_start: Compile time:     Apr 20 2021 20:25:26
I (1074) cpu_start: ELF file SHA256:  d73c6b4d551ca241...
I (1080) cpu_start: ESP-IDF:          v4.4-dev-744-g1cb31e509
I (1086) heap_init: Initializing. RAM available for dynamic allocation:
I (1093) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
I (1100) heap_init: At 3FFC7EE0 len 00018120 (96 KiB): DRAM
I (1106) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1112) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1119) heap_init: At 4009838C len 00007C74 (31 KiB): IRAM
I (1125) spiram: Adding pool of 4084K of external SPI memory to heap allocator
I (1134) spi_flash: detected chip: generic
I (1138) spi_flash: flash io: qio
I (1150) cpu_start: Starting scheduler on PRO CPU.
[... 62 lines cut ...]
Got GET request!
RQ: action=addUser&userName=1tfd7ot4veqj2hlcl196z8atv&blob=4wxXZZrgfOYU6FezeK0MFWo68ExTkCKUmzaADZfUx3w%2FhPphvtXYBeWCeQhuSoJVVCmb1
Got POST request!
Connecting with AP <guc3-accesspoint-e-fgwz.ap.spotify.com:4070>

abort() was called at PC 0x401710ef on core 0

Backtrace:0x4008de65:0x3fff48d0 0x4008e5ed:0x3fff48f0 0x40094259:0x3fff4910 0x401710ef:0x3fff4980 0x40171136:0x3fff49a0 0x40160e40

ELF file SHA256: d73c6b4d551ca241

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7052
ho 0 tail 12 room 4
load:0x40078000,len:15404
load:0x40080400,len:4452
entry 0x40080694

```

hajdbo, 2021-04-23:
Oh. I hoped it was due to a lack of external ram on my WROOM, but your chip has 8Mb spiram and has the same problem it seems.

geoffreyheller, 2021-04-23:
I am wondering if perhaps my idf menuconfig settings are different than what you are using, because your logs seem to indicate that your ESP32 made it farther than mine.

After changing the sink in main.cpp, I'm doing a very simple `idf.py build ` followed by `idf.py -p /dev/ttyUSB0 flash`.   

Note that I am building and flashing from a Ubuntu 20.04 running in VirtualBox.  The device I am attempting to connect with is an iPhone 11.  Not sure if all that makes any difference.

feelfreelinux, 2021-04-23:
> I am having a similar issue. cspot is visible on Spotify but when I try to connect I get an abort().
> 
> In my case I am using an ESP32-WROVER-IB, which is mounted on @fgervais's breakout board: https://github.com/fgervais/esp32-wrover-breakout
> 
> I am also using a PCM5102, so I am wondering if that might somehow be a factor. I implemented my audiosink declaration in main.cpp the same way that Boris did.
> 
> My log looks a bit different than Boris's. I'm assuming he has additional logging turned on somewhere:
> 
> ```
> Brown�ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0030,len:7052
> ho 0 tail 12 room 4
> load:0x40078000,len:15404
> load:0x40080400,len:4452
> entry 0x40080694
> I (29) boot: ESP-IDF v4.4-dev-744-g1cb31e509 2nd stage bootloader
> I (29) boot: compile time 20:44:45
> I (29) boot: chip revision: 1
> I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
> I (40) qio_mode: Enabling default flash chip QIO
> I (45) boot.esp32: SPI Speed      : 80MHz
> I (50) boot.esp32: SPI Mode       : QIO
> I (54) boot.esp32: SPI Flash Size : 4MB
> I (59) boot: Enabling RNG early entropy source...
> I (64) boot: Partition Table:
> I (68) boot: ## Label            Usage          Type ST Offset   Length
> I (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (90) boot:  2 factory          factory app      00 00 00010000 00300000
> I (98) boot: End of partition table
> I (102) boot_comm: chip revision: 1, min. application chip revision: 0
> I (109) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=3f430h (259120) map
> I (195) esp_image: segment 1: paddr=0004f458 vaddr=3ffbdb60 size=00bc0h (  3008) load
> I (197) esp_image: segment 2: paddr=00050020 vaddr=400d0020 size=d30d8h (864472) map
> I (460) esp_image: segment 3: paddr=00123100 vaddr=3ffbe720 size=04260h ( 16992) load
> I (466) esp_image: segment 4: paddr=00127368 vaddr=40080000 size=1838ch ( 99212) load
> I (500) esp_image: segment 5: paddr=0013f6fc vaddr=50000000 size=00010h (    16) load
> I (514) boot: Loaded app from partition at offset 0x10000
> I (514) boot: Disabling RNG early entropy source...
> I (526) psram: This chip is ESP32-D0WD
> I (526) spiram: Found 64MBit SPI RAM device
> I (526) spiram: SPI RAM mode: flash 80m sram 80m
> I (531) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
> I (539) cpu_start: Pro cpu up.
> I (542) cpu_start: Starting app cpu, entry point is 0x40081710
> I (0) cpu_start: App cpu up.
> I (1045) spiram: SPI SRAM memory test OK
> I (1053) cpu_start: Pro cpu start user code
> I (1054) cpu_start: cpu freq: 240000000
> I (1054) cpu_start: Application information:
> I (1057) cpu_start: Project name:     cspot-esp32
> I (1062) cpu_start: App version:      c414cd6-dirty
> I (1068) cpu_start: Compile time:     Apr 20 2021 20:25:26
> I (1074) cpu_start: ELF file SHA256:  d73c6b4d551ca241...
> I (1080) cpu_start: ESP-IDF:          v4.4-dev-744-g1cb31e509
> I (1086) heap_init: Initializing. RAM available for dynamic allocation:
> I (1093) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
> I (1100) heap_init: At 3FFC7EE0 len 00018120 (96 KiB): DRAM
> I (1106) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
> I (1112) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
> I (1119) heap_init: At 4009838C len 00007C74 (31 KiB): IRAM
> I (1125) spiram: Adding pool of 4084K of external SPI memory to heap allocator
> I (1134) spi_flash: detected chip: generic
> I (1138) spi_flash: flash io: qio
> I (1150) cpu_start: Starting scheduler on PRO CPU.
[... 65 lines cut ...]
> Connecting with AP <guc3-accesspoint-e-fgwz.ap.spotify.com:4070>
> 
> abort() was called at PC 0x401710ef on core 0
> 
> Backtrace:0x4008de65:0x3fff48d0 0x4008e5ed:0x3fff48f0 0x40094259:0x3fff4910 0x401710ef:0x3fff4980 0x40171136:0x3fff49a0 0x40160e40
> 
> 
> ELF file SHA256: d73c6b4d551ca241
> 
> Rebooting...
> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0030,len:7052
> ho 0 tail 12 room 4
> load:0x40078000,len:15404
> load:0x40080400,len:4452
> entry 0x40080694
> ```

Looks like you haven't enabled PSRAM in idf.py menuconfig, correct logs should show something like this:
[image]

feelfreelinux, 2021-04-23:
> Oh. I hoped it was due to a lack of external ram on my WROOM, but your chip has 8Mb spiram and has the same problem it seems.

Unfortunately its impossible to run cspot on a device without external PSRAM, a wrover chip is required for it to run.

geoffreyheller, 2021-04-23:
For the ESP32-WROVER-IB abort() that I am seeing, is there some additional debugging I can turn on to help diagnose the issue?

feelfreelinux, 2021-04-23:
> For the ESP32-WROVER-IB abort() that I am seeing, is there some additional debugging I can turn on to help diagnose the issue?

make sure you see the "Adding pool of ..." log when booting up esp32. I'll try to add better logging soon :)

geoffreyheller, 2021-04-25:
I'm interested in why in @hajdbo 's logs his system seems to get farther along than mine:

@hajdbo 's log snippet:
```
I (3170) TAG: Connected to AP, start spotify receiver
Starting zeroconf auth server at port 11459
RQ:
Got GET request!
RQ:
Got GET request!
RQ:
Got GET request!
RQ: action=addUser&userName=zboubi&blob=XXXXX&deviceName=CORSAIRONE&deviceId=XXXXXX&version=2.7.1
Got POST request!
Connecting with AP <gae2-accesspoint-e-qf6p.ap.spotify.com:4070>
Connected to spotify server
Sending APHello packet...
recv
Maybe corn?
Received shannon keys
Authorization successful
Got ping, syncing timestamp
I (56320) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=8
I (56340) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
SUB
Received packet with code 2 of length 336
Received packet with code 118 of length 2
Received packet with code 27 of length 2
Received country code

```

My log snippet from the same point:
```
I (6200) TAG: Connected to AP, start spotify receiver
Starting zeroconf auth server at port 33873
RQ: 
Got GET request!
RQ: 
Got GET request!
RQ: 
Got GET request!
RQ: 
Got GET request!
RQ: 
Got GET request!
RQ: 
Got GET request!
RQ: 
Got GET request!
RQ: action=addUser&userName=1tfd7ot4veqj2hlcl196z8atv&blob=nppWCo%2FC4OuQpuaKrvoY8HWNQTUkBrWo703dEr1ah6GiTCgZHqDb3PktFIaATvDYcXA1
Got POST request!
Connecting with AP <guc3-accesspoint-e-8fth.ap.spotify.com:4070>

abort() was called at PC 0x40171063 on core 0

Backtrace:0x4008ddc5:0x3fff59f0 0x4008e531:0x3fff5a10 0x4009419d:0x3fff5a30 0x40171063:0x3fff5aa0 0x401710aa:0x3fff5ac0 0x40160d0

ELF file SHA256: ca115f099bb5ee80

Rebooting...
```
You can see that my device never makes it to the line "Connected to spotify server".  

This makes me wonder if perhaps I have something misconfigured in my build or perhaps menuconfig.   Would you mind sharing a few details about how you are building and what config steps you are taking?

hajdbo, 2021-04-27:
@geoffreyheller in menuconfig, enable Component config --> ESP32-specific --> Support for external,, SPI-connected RAM.

closed 2021-04-27 by hajdbo

nezen, 2021-11-17:
@feelfreelinux greetings, do you think it's possible to run cspot on an esp32 board with 4 MB psram?

nezen, 2021-11-17:
Sorry, btw is this a reversed engineering project? I didn't find any spotify connect documents on their website.

feelfreelinux, 2021-11-17:
> @feelfreelinux greetings, do you think it's possible to run cspot on an esp32 board with 4 MB psram?

Yes, 4mb is enough

feelfreelinux, 2021-11-17:
> Sorry, btw is this a reversed engineering project? I didn't find any spotify connect documents on their website.

And yes, the API is not public, this is all based on reverse engineering and the work done on librespot projects