feelfreelinux/cspot #53: Does PCM5102 also work with MAX98357

don41382, 2021-04-23:
Hi Filip,

Thanks for the awesome project! Does the PCM5102 sink also works with the MAX98357? Maybe somebody already checked this out.

Felix

feelfreelinux, 2021-04-26:
Hi! From what I've seen the I2S format required for it is pretty similar, so it should work without any issues. Let me know if it works fine, we can always just add support for MAX98357 separately :)

don41382, 2021-04-26:
I'll try and let you know. My biggest challenge is still to get it deployed via platformio. I am coming more from the Arduino world and not so familiar with working directly with esp-idf. This is off-topic, but do you have some guidance to deploy it to a esp32?

don41382, 2021-07-01:
Finally got my wrover TTGO T7 working and I can confirm, that the `PCM5102AudioSink` works also with the `MAX98357A`.

I just had to change the pins according to my connections:

```
    i2s_pin_config_t pin_config = {
        .bck_io_num = 26,
        .ws_io_num = 25,
        .data_out_num = 22,
        .data_in_num = -1 //Not used
    };
```

closed 2021-07-01 by don41382