feelfreelinux/cspot #56: cpu_start: Failed to init external RAM

don41382, 2021-04-28:
Hi guys,

When I spot the last commit "Added platformio support" I got excited and had to try it out. Building & compiling seems to work in platformio. Thanks for that.

But my `Lolin32` board is stucked in a endless boot loop.

* Board: Lolin32
* Chip: ESP32-D0WD
* 4 MB FLASH PSRAM

```rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7676
load:0x40078000,len:14368
load:0x40080400,len:5360
entry 0x40080710
␛[0;32mI (29) boot: ESP-IDF 3.40200.210118 2nd stage bootloader␛[0m
␛[0;32mI (29) boot: compile time 16:36:00␛[0m
␛[0;32mI (29) boot: chip revision: 1␛[0m
␛[0;32mI (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0␛[0m
␛[0;32mI (40) qio_mode: Enabling default flash chip QIO␛[0m
␛[0;32mI (45) boot.esp32: SPI Speed      : 40MHz␛[0m
␛[0;32mI (50) boot.esp32: SPI Mode       : QIO␛[0m
␛[0;32mI (54) boot.esp32: SPI Flash Size : 4MB␛[0m
␛[0;32mI (59) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (64) boot: Partition Table:␛[0m
␛[0;32mI (68) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (82) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (90) boot:  2 factory          factory app      00 00 00010000 00300000␛[0m
␛[0;32mI (97) boot: End of partition table␛[0m
␛[0;32mI (102) boot_comm: chip revision: 1, min. application chip revision: 0␛[0m
␛[0;32mI (109) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x40768 (264040) map␛[0m
␛[0;32mI (214) esp_image: segment 1: paddr=0x00050790 vaddr=0x3ffbdb60 size=0x0416c ( 16748) load␛[0m
␛[0;32mI (221) esp_image: segment 2: paddr=0x00054904 vaddr=0x40080000 size=0x00404 (  1028) load␛[0m
␛[0;32mI (222) esp_image: segment 3: paddr=0x00054d10 vaddr=0x40080404 size=0x0b308 ( 45832) load␛[0m
␛[0;32mI (249) esp_image: segment 4: paddr=0x00060020 vaddr=0x400d0020 size=0xd2b28 (863016) map␛[0m
␛[0;32mI (564) esp_image: segment 5: paddr=0x00132b50 vaddr=0x4008b70c size=0x0c3e8 ( 50152) load␛[0m
␛[0;32mI (599) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (599) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (599) psram: This chip is ESP32-D0WD␛[0m
␛[0;31mE (604) psram: PSRAM ID read error: 0xffffffff␛[0m
␛[0;31mE (609) cpu_start: Failed to init external RAM, needed for external .bss segment␛[0m
Re-enable cpu cache.

abort() was called at PC 0x40081ed0 on core 0````

Any idea why it can't access the RAM?

feelfreelinux, 2021-04-28:
IIRC, Lolin32 uses a "WROOM" chip - it simply doesn't have enough ram. "4MB Flash PSRAM" is probably a misleading description provided by the chip's seller, as the chip doesn't have 4MB PSRAM (it's just flash memory).

don41382, 2021-04-29:
Thanks for quick clarification. I guess, I was falling for the sellers description.  The FireBeetle should work right, it has 520KB PSRAM?

https://www.dfrobot.com/product-1590.html

don41382, 2021-04-29:
Do you think the `LILYGO®TTGO T7 V 1,5` would work? It's a ESP32 ESP32-WROVER and 8MB of PSRAM - if I can trust the sellers description.

https://de.aliexpress.com/item/32977375539.html

feelfreelinux, 2021-04-29:
> Do you think the `LILYGO®TTGO T7 V 1,5` would work? It's a ESP32 ESP32-WROVER and 8MB of PSRAM - if I can trust the sellers description.
> 
> https://de.aliexpress.com/item/32977375539.html

They work well - I've used them with cspot before :) I'm surprised with the price though, I got them for around 5$ last summer

don41382, 2021-07-19:
I am closing the issue, as the README quotes, only `WROVER` is currently supported. 

And I can confirm, that the `LILYGO®TTGO T7 V 1,5` does the job.

closed 2021-07-19 by don41382