feelfreelinux/cspot #58: getting core dumped when it start playin on Linux/cli

mehmetugursoy, 2021-05-18:
I compiled app successfully  under ubuntu 18.04 also did same thing on raspberry pi. 

on Both OS , i got core dumped , in same place.  

`mehmet@ubuntu-vm:~/cspot/targets/cli/build$ ./cspotcli 
I ZeroconfAuthenticator.cpp:23: Starting zeroconf auth server at port 7687
*** WARNING *** The program 'cspotcli' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
D ZeroconfAuthenticator.cpp:101: RQ: 
I ZeroconfAuthenticator.cpp:132: Got GET request!
D ZeroconfAuthenticator.cpp:101: RQ: 
I ZeroconfAuthenticator.cpp:132: Got GET request!
D ZeroconfAuthenticator.cpp:101: RQ: 
I ZeroconfAuthenticator.cpp:132: Got GET request!
D ZeroconfAuthenticator.cpp:101: RQ: action=addUser&userName=11139653616&blob=yVCJnI3Br3rCl8g%2BPgAwwsziwzQOXYvyPWn4Au%2FepzwaEt6owWFo0uxyV8rTB6fTL525gMed24D89CKmhcKLXERSJsvorGGlmmJJ5QoT7QHCP31bvxgoc%2FKNknjBf4K0MqN5Dwii8O2GHqWo%2FogGfztIRryh%2BAgZqytB4WFhNOQoXp1NUk2ZRGqIBeGvv1BgCYNTEbtkeoNByf%2BExzTCjqqeN%2FvO2RgtEHp%2BZuzme0Kf0hmGk26YxAjX7v%2B7T5NcoMtxaZInw3xBFY01rX%2FJydYyoVOAgZxz3Dxa6xxpwHo6DTfTOagouBz0EwsPrUouwxP9HbuEw4vGRvSo&clientKey=33KSRqQOvxu3dXHujVMvjPjwhF9Obe7N%2BKSUmcKzU8AUs5%2Fd64SkRcuMUo4klmNCnzlOweQKWrg6gRsq4ZKtunQGmr1ZHsuWaz%2B16abanGM91X5H9olGfZ3XE%2BeH4Vgk&tokenType=default&loginId=1e0b98f40cb5817266342dcc0dbffe46&deviceName=5CG93280PV&deviceId=468002956a218a6d733510211a2ec88a25d8457b&version=2.7.1
I ZeroconfAuthenticator.cpp:106: Got POST request!
D Session.cpp:27: Connecting with AP <gew1-accesspoint-e-tq43.ap.spotify.com:4070>
D PlainConnection.cpp:67: Connected to spotify server
D Session.cpp:30: Sending APHello packet...
D Session.cpp:80: Processing AP hello response...
D Session.cpp:82: Received AP hello response
D Session.cpp:123: Received shannon keys
D Session.cpp:59: Authorization successful
D MercuryManager.cpp:168: Got ping, syncing timestamp
PCM name: 'default'
PCM state: PREPARED
channels: 2 (stereo)
period_time = 799
required buff_size: 140
D MercuryManager.cpp:267: executing MercuryType SUB
D MercuryManager.cpp:196: Received packet with code 2 of length 336
D MercuryManager.cpp:196: Received packet with code 118 of length 2
D MercuryManager.cpp:196: Received packet with code 27 of length 2
D MercuryManager.cpp:203: Received country code: TR
D MercuryManager.cpp:196: Received packet with code 80 of length 6087
D MercuryManager.cpp:196: Received packet with code 105 of length 0
D MercuryManager.cpp:196: Received packet with code 31 of length 17
D MercuryManager.cpp:196: Received packet with code 74 of length 4
D MercuryManager.cpp:196: Received packet with code 179 of length 51
D MercuryManager.cpp:267: executing MercuryType SEND
D SpircController.cpp:28: Sent kMessageTypeHello!
D MercuryManager.cpp:196: Received packet with code 178 of length 51
D MercuryManager.cpp:196: Received packet with code 181 of length 414
D SpircController.cpp:45: Notify frame
D MercuryManager.cpp:196: Received packet with code 181 of length 330
D SpircController.cpp:45: Notify frame
D MercuryManager.cpp:196: Received packet with code 181 of length 496
D SpircController.cpp:45: Notify frame
D MercuryManager.cpp:196: Received packet with code 181 of length 619
D SpircController.cpp:100: Load frame!
I PlayerState.cpp:113: ---- Track count 0
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
Aborted (core dumped)
`
Could you please give some idea or show some hint to pass this.  

My Actual aim is  to implement vs1053(http://www.vlsi.fi/en/products/vs1005.html) as a vorbis  hardware code. i tested it with librespot and it works. try to do same thing with yours.

feelfreelinux, 2021-05-19:
Hi! Can you tell me which track / album / playlist you're trying to play?

feelfreelinux, 2022-12-18:
Closed due to inactivity

closed 2022-12-18 by feelfreelinux