feelfreelinux/cspot #66: What do we need to make cspot an autonomous client?

hajdbo, 2021-06-28:
Right now cspot is setup as a *sink*, to be controlled through the app on a phone or computer.

I'm considering building an autonomous jukebox, which would trigger songs/albums based on NFC cards (e.g. the card would contain the spotify album_id).

Would it be a major effort to support being a lightweight client that supports login/pass authentication + launching a specific song or album?

feelfreelinux, 2021-06-28:
Hi! Actually, cspot does currently support everything needed for this. Login / Pass authentication is already in place (see targets/cli), and after exposing a function in the player that allows you to take a track id as playback, you should be able to achieve your goal

feelfreelinux, 2021-06-28:
Well, album playback by id is missing, but that's something rather easy to add

hajdbo, 2021-06-28:
I did not realize it was that close. I'll work on a branch and try to make it work. Thx!

closed 2021-06-28 by hajdbo

hillstub, 2022-08-21:
Really appreciate the work you did to make this work on the ESP32!! I also would like to connect an NFC reader to an ESP32 so I can play an album/song based on the card that's read. 

Could you help me a bit on what approach to take to extend the player to take a song or album uri and start playing?