feelfreelinux/cspot #68: ***ERROR*** A stack overflow in task cspot has been detected.

don41382, 2021-07-18:
TTGO T7 
Chip: ESP32-D0WD

Sometimes when I switch tracks or playing for a long time the ESP32 dies with a "Stackoverflow".

```
I PlayerState.cpp:114: ---- Track count 49
D MercuryManager.cpp:267: executing MercuryType GET
D MercuryManager.cpp:267: executing MercuryType SEND
D MercuryManager.cpp:196: Received packet with code 178 of length 1431
D MercuryManager.cpp:234:  MercuryType::UNSUB response->parts[0].size() = 1200
I SpotifyTrack.cpp:76: Track name: Ich frag mich - 4k Remix
D SpotifyTrack.cpp:77: trackInfo.restriction.size() = 1
D MercuryManager.cpp:196: Received packet with code 178 of length 50
D MercuryManager.cpp:196: Received packet with code 13 of length 20
I SpotifyTrack.cpp:132: Successfully got audio key!
D AudioChunkManager.cpp:15: Chunk requested 0
D AudioChunkManager.cpp:78: ID: 0: header decrypt!
D AudioChunkManager.cpp:15: Chunk requested 1
D AudioChunkManager.cpp:90: ID: 0: Starting decrypt!
D AudioChunkManager.cpp:78: ID: 1: header decrypt!
D AudioChunkManager.cpp:90: ID: 1: Starting decrypt!
D MercuryManager.cpp:267: executing MercuryType SEND
D ChunkedAudioStream.cpp:309: Change in current chunk 0

***ERROR*** A stack overflow in task cspot has been detected.

Backtrace:0x4008e6a1:0x3ffd47d0 0x4008ee21:0x3ffd47f0 0x400924a5:0x3ffd4810 0x40090ae7:0x3ffd4890 0x4008ef18:0x3ffd48d0 0x4008eeca:0x00000000 |<-CORRUPTED
0x4008e6a1: panic_abort at dev/iot/esp/esp-idf/components/esp_system/panic.c:367

0x4008ee21: esp_system_abort at dev/iot/esp/esp-idf/components/esp_system/system_api.c:112

0x400924a5: vApplicationStackOverflowHook at dev/iot/esp/esp-idf/components/freertos/port/xtensa/port.c:489

0x40090ae7: vTaskSwitchContext at dev/iot/esp/esp-idf/components/freertos/tasks.c:3289

0x4008ef18: _frxt_dispatch at dev/iot/esp/esp-idf/components/freertos/port/xtensa/portasm.S:432

0x4008eeca: _frxt_int_exit at dev/iot/esp/esp-idf/components/freertos/port/xtensa/portasm.S:231

ELF file SHA256: 95431bec19368daa

Rebooting...
ets Jun  8 2016 00:22:57
```

Any idea why this is happening?

don41382, 2021-07-19:
I found the responsible crashing task, which is the `MercuryManager`. I first thought it's the main task of the `main.cpp`

``` auto taskHandle = xTaskCreatePinnedToCore(&cspotTask, "cspot", 8192 * 8, NULL, 5, NULL, 0); ```

But the name "cspot" was just inheritance by the `MercuryManager`. So I added a name to the created task in the `Task.h` and increased the `stack_size`.

```
#ifdef ESP_PLATFORM
      esp_pthread_cfg_t cfg = esp_pthread_get_default_config();
      cfg.stack_size = (12 * 1024);
      cfg.inherit_cfg = true;
      cfg.pin_to_core = 1;
      cfg.thread_name = "cspotMercury";
      esp_pthread_set_cfg(&cfg);
#endif
```

Now it works. No crashes. But how do you determine a good / reasonable `stack_size`? Just by error and trial?

feelfreelinux, 2021-07-24:
> At first I thought the overflowing task is the one in the `main.cpp`, then I thought it's the `MercuryManagerTask` ....
> 
> Now I figured out, that the `Task.h` is not only inherit from the `MercuryManager` but also from
> 
> * `AudioChunkManager.h`
> * `Player.h`
> 
> So all of them share the same `stack_size` of `(8 * 1024)` and inherited the name of the `main.cpp` task - because no other was defined. That got me really confused.
> 
> Task.h
> 
> ```
> #ifdef ESP_PLATFORM
>       esp_pthread_cfg_t cfg = esp_pthread_get_default_config();
>       cfg.stack_size = (8 * 1024);
>       cfg.inherit_cfg = true;
>       cfg.pin_to_core = 1;
>       esp_pthread_set_cfg(&cfg);
> #endif
> ```
> 
> I enforced `Task.h` to accept `stack_size` and `thread_name` as construction parameters and defined them for each implementing class. Afterwards I figured out that the `Player.h` task stack with `8192` is too small and kept throwing the stackoverflow error. I increased it by `9216`. Now it works.
> 
> Two questions:
> 
> a) How do you determine a good / reasonable `stack_size`? Just by error and trial?
> b) Should I create a PR for my changes?

Hi, sorry for the late reply. Right now I set the stack_size by basically trial and error, so I'd gladly accept a PR that allows to set those parameters separately.

feelfreelinux, 2021-12-22:
Resolved by moving to bell on newest master, Stack size is separately adjusted + values were verified

closed 2021-12-22 by feelfreelinux