feelfreelinux/cspot #76: Player commands/Status

philippe44, 2021-11-13:
Hi - I've started to integrated cspot in https://github.com/sle118/squeezelite-esp32 and it's starting to work so far, but one thing I'm missing is a player's status/command when playback is started, stopped, paused.... I can add a few methods to the audiosink and call them in Player.cpp, but it seems that spotify does not always send this stop/play event, especially when playback is already on one device and track is just "transferred"

feelfreelinux, 2021-11-14:
Hi! Great to see you implementing it in that project! I'm currently working on integrating cspot with my own audio player software on esp32, and I'm going to fix the Player commands / events.
About stack sizes: It's currently quite a mess, I'm going to soon release a refactor that will make the stack sizes configurable per-task. The few dropped frames are most likely a bug in ChunkedAudioStream, that does not buffer properly at the beginning of a song. Probably preloading beginning chunks moment before they're requested would fix the problem.

philippe44, 2021-11-14:
I've changed the stack sizes according to what I've measured and see how it goes (there are stats in squeezeESP32). 

Re AudioChunks, any hints where I should look?

Something else I forgot to ask: do you gather metadata and is there a way to get them (I have a callback mechanism to display these) and is there a way to feedback commands to the Spotify controller as well (like volume/play/pause/stop)

philippe44, 2021-11-14:
BTW, it's really good to have the choice of storage backend being left to user and not in the core because, for example, I don't want to use spiffs, so the read/write file is a simple nvs implementation (for config - for auth I'm not storing anything so far)

feelfreelinux, 2021-12-22:
Resolved, events and player commands are now handled through SpircController:
https://github.com/feelfreelinux/cspot/blob/master/cspot/include/SpircController.h#L55

closed 2021-12-22 by feelfreelinux