feelfreelinux/cspot #77: Authorization declined

Timvrakas, 2021-11-18:
I got CLI and ESP32 building on MacOS.
```
timv@shafer build % ./cspotcli
D Session.cpp:27: Connecting with AP <guc3-accesspoint-a-h3c6.ap.spotify.com:4070>
D PlainConnection.cpp:67: Connected to spotify server
D Session.cpp:30: Sending APHello packet...
D Session.cpp:80: Processing AP hello response...
D Session.cpp:82: Received AP hello response
D Session.cpp:123: Received shannon keys
E Session.cpp:68: Authorization declined
```

What can I look at to help debug? I am using a premium account.

feelfreelinux, 2021-11-19:
What version of MacOS have you ran cspot on? This is very interesting, considering that the esp32 has authorized successfully (https://github.com/feelfreelinux/cspot/issues/78)

Could you try to delete `authBlob.json` and try again?

Timvrakas, 2021-11-19:
This is MacOS Monterey (12.0.1), running on a new M1 MacBookPro, (IE: ARM64). I hadn't though about it but that is a likely culprit.
I've tried removing authBlob.json and retrying, always the same result.

zoharcochavi, 2021-11-28:
I'm not sure if this is relevant, but I get exactly the same error when running on the ESP32.

kuba2k2, 2021-12-08:
Same here, on the ESP32. As I have already worked with the low-level protocols used here some time ago, I will try to find out what the problem is.

feelfreelinux, 2021-12-08:
> Same here, on the ESP32. As I have already worked with the low-level protocols used here some time ago, I will try to find out what the problem is.

Thanks a lot, I can't recreate the issue myself. Feel free to reach out to me via telegram or discord, I'd love to explain a bit of internals :)

kuba2k2, 2021-12-08:
Okay, the problem is in the build info sent with ClientHello, containing `PRODUCT_PARTNER` which does not work for free accounts. This means it's not really an issue, just like #71. It could be helpful to add a message to the `Authorization declined` error, informing about the possible source of the problem, being a free account.

zoharcochavi, 2021-12-08:
Oh damn, that's interesting. I'm not using a free account tho. Does that refer to the account used for the person wanting to connect to the ESP? Or am I missing some auth config that have to be set during build?

kuba2k2, 2021-12-08:
This refers to any authenticated account, such as the one being connected (transferred) to the ESP (from e.g. a smartphone).

Timvrakas, 2021-12-10:
I've had success with the same phone and laptop (logged in with the same user) for the ESP32 target work every time, but the auth fails every time on MacOS aarch64.

mixsoupe, 2022-05-17:
I have the same "Authorization declined" issue on the ESP32 with a Spotify premium account. I tried to connect with an Android phone and with Windows 10 with no luck. Have you some updates on this ?

feelfreelinux, 2022-05-17:
most likely solved in https://github.com/feelfreelinux/cspot/commit/27ceb08352a0ab628c6a772a4a9cbb8e00ef210a

cspot requires a premium account to run, so the initial bug with authentication on free accounts still remain. In order not to interfere with Spotify's TOS, I do not plan to support free accounts (even if ads support was added).

markuspi, 2022-05-29:
I'm still getting an "Authorization declined" error when using `cspotcli` with zeroconf authorization on Kubuntu 22.04:

```
$ ./cspotcli
*** WARNING *** The program 'cspotcli' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
I ZeroconfAuthenticator.cpp:55: Zeroconf registering handlers
I HTTPServer.cpp:68: Starting server at port 2137
I ZeroconfAuthenticator.cpp:26: Got request for info
I ZeroconfAuthenticator.cpp:26: Got request for info
I ZeroconfAuthenticator.cpp:26: Got request for info
I ZeroconfAuthenticator.cpp:37: Got request for adding user
I main.cpp:80: Creating player
D Session.cpp:39: Connecting with AP <ap-gew1.spotify.com:4070>
D PlainConnection.cpp:70: Connected to spotify server
D Session.cpp:42: Sending APHello packet...
D Session.cpp:102: Processing AP hello response...
D Session.cpp:104: Received AP hello response
D Session.cpp:150: Received shannon keys
E Session.cpp:90: Authorization declined
I PlainConnection.cpp:175: Closing socket...
```
I'm using a Spotify premium account. Zeroconf works still works fine with my account on librespot-java for example.

When I try username/password authentication, I get an "authorization successful" message, but the client does not show up in other spotify devices, so playing music this way also seems broken for me:
```
$ ./cspotcli -u <username> -p <password>
I main.cpp:80: Creating player
D Session.cpp:39: Connecting with AP <ap-gew1.spotify.com:4070>
D PlainConnection.cpp:70: Connected to spotify server
D Session.cpp:42: Sending APHello packet...
D Session.cpp:104: Processing AP hello response...
D Session.cpp:106: Received AP hello response
D Session.cpp:152: Received shannon keys
I Session.cpp:54: Username: spotify@markuspi.de AuthType: 0
D Session.cpp:83: Authorization successful
Start
stop
D MercuryManager.cpp:299: executing MercuryType SUB
D MercuryManager.cpp:216: Received packet with code 2 of length 336
D MercuryManager.cpp:216: Received packet with code 118 of length 2
D MercuryManager.cpp:216: Received packet with code 27 of length 2
D MercuryManager.cpp:223: Received country code: DE
D MercuryManager.cpp:216: Received packet with code 80 of length 6485
D MercuryManager.cpp:216: Received packet with code 105 of length 0
D MercuryManager.cpp:216: Received packet with code 31 of length 17
D MercuryManager.cpp:216: Received packet with code 181 of length 513
Decode failed: string overflow
D MercuryManager.cpp:216: Received packet with code 74 of length 4
D MercuryManager.cpp:216: Received packet with code 179 of length 59
D MercuryManager.cpp:299: executing MercuryType SEND
D SpircController.cpp:32: Sent kMessageTypeHello!
D MercuryManager.cpp:216: Received packet with code 178 of length 188
D MercuryManager.cpp:254:  MercuryType::UNSUB response->parts[0].size() = 127
```