feelfreelinux/cspot #79: Codec Development for ES8311 (LyraT-Mini)

Timvrakas, 2021-11-27:
I have a LyraT-Mini V1.2, for which I can develop working examples from https://github.com/espressif/esp-adf
I'd like to get it cspot working, but I haven't worked with audio codecs or I2S before, so I'm not sure where to start.

The board/pin definitions are [here](https://github.com/espressif/esp-adf/blob/master/components/audio_board/lyrat_mini_v1_1/board_pins_config.c)
The chip-level driver is [here](https://github.com/espressif/esp-adf/blob/master/components/audio_hal/driver/es8311/es8311.c)

I'm not sure how to integrate this with cspot. If these libraries are functional, should I try using them directly, and use the existing BufferedAudioSink -> i2s_write() that exists?

I changed the AC101 pins and i2c address, and the board puts out a soft clicking noise, but no meaningful sound.

Timvrakas, 2021-11-27:
See #80

feelfreelinux, 2021-11-27:
Hi! Thanks a lot for developing this further! You should implement the driver by adding the https://github.com/espressif/esp-adf/blob/master/components/audio_hal/driver/es8311/es8311.c driver into cspot / targets / esp32 and call proper initialization and volume change methods from a new audio sink. See `AC101AudioSink` for reference. About the clicking sound after each chunks - I'm currently working on it, please disable MBEDTLS hardware acceleration to fix it for yourself too :)

Feel free to reach out to me on telegram / discord / matrix (ESP32_Audio channel). I'd love to help out!