feelfreelinux/cspot #85: esp32 cpu resetting with the new ESP-IDF version 4.3.2

ehabulhaq, 2021-12-21:
Hi,

I have updated my esp idf version to the latest release of version 4.3.2. They have updated the xtensa tools as well with the new release. 

Below is the log output of the chip when I play a song over Spotify:

E (47398) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (47398) task_wdt:  - IDLE (CPU 1)
E (47398) task_wdt: Tasks currently running:
E (47398) task_wdt: CPU 0: IDLE
E (47398) task_wdt: CPU 1: cspot
E (47398) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4012D032:0x3FFBE9E0 0x400844F9:0x3FFBEA00 0x4019B3EF:0x3FFB5E40 0x400D83B5:0x3FFB5E60 0x40090288:0x3FFB5E80
0x4012d032: task_wdt_isr at C:/Users/EHaq/esp/esp-idf/components/esp_common/src/task_wdt.c:189

0x400844f9: _xt_lowint1 at C:/Users/EHaq/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4019b3ef: cpu_ll_waiti at C:/Users/EHaq/esp/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at C:/Users/EHaq/esp/esp-idf/components/esp_pm/pm_impl.c:827

0x400d83b5: esp_vApplicationIdleHook at C:/Users/EHaq/esp/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x40090288: prvIdleTask at C:/Users/EHaq/esp/esp-idf/components/freertos/tasks.c:3846

E (47398) task_wdt: Print CPU 1 backtrace

Backtrace:0x40086829:0x3FFBEFE0 0x400844F9:0x3FFBF000 0x40100911:0x3FFD07E0 0x400FF509:0x3FFD0830 0x400FF8D5:0x3FFD08E0 0x400FDA87:0x3FFD0A70 0x400FB840:0x3FFD0AA0 0x400FCB72:0x3FFD0B00 0x400FAAAB:0x3FFD0B30 0x400F71D3:0x3FFD0BA0 0x40199BEB:0x3FFD0BD0
0x400D2E80:0x3FFD0BF0
0x40086829: esp_crosscore_isr at C:/Users/EHaq/esp/esp-idf/components/esp32/crosscore_int.c:77

0x400844f9: _xt_lowint1 at C:/Users/EHaq/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x40100911: vorbis_book_decodevv_add at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/tremor/codebook.c:787

0x400ff509: res_inverse at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/tremor/res012.c:210

0x400ff8d5: mapping_inverse at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/tremor/mapping0.c:171 (discriminator 2)

0x400fda87: vorbis_dsp_synthesis at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/tremor/dsp.c:212

0x400fb840: _fetch_and_process_packet at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/tremor/vorbisfile.c:556

0x400fcb72: ov_read at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/tremor/vorbisfile.c:1581

0x400faaab: ChunkedAudioStream::startPlaybackLoop() at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/src/ChunkedAudioStream.cpp:105

0x400f71d3: Player::runTask() at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/src/Player.cpp:79

0x40199beb: Task::taskEntryFunc(void*) at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/include/Task.h:41

0x400d2e80: pthread_task_func at C:/Users/EHaq/esp/esp-idf/components/pthread/pthread.c:209 (discriminator 15)

D AudioChunkManager.cpp:15: Chunk requested 28
D ChunkedAudioStream.cpp:259: Chunk req end pos 4194304
D AudioChunkManager.cpp:78: ID: 28: header decrypt!
D AudioChunkManager.cpp:90: ID: 28: Starting decrypt!
D AudioChunkManager.cpp:15: Chunk requested 29
D ChunkedAudioStream.cpp:259: Chunk req end pos 4325376
D AudioChunkManager.cpp:78: ID: 29: header decrypt!
D AudioChunkManager.cpp:90: ID: 29: Starting decrypt!
D AudioChunkManager.cpp:15: Chunk requested 30
D ChunkedAudioStream.cpp:259: Chunk req end pos 4456448
D AudioChunkManager.cpp:78: ID: 30: header decrypt!
D AudioChunkManager.cpp:90: ID: 30: Starting decrypt!
E (52398) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (52398) task_wdt:  - IDLE (CPU 1)
E (52398) task_wdt: Tasks currently running:
E (52398) task_wdt: CPU 0: IDLE
E (52398) task_wdt: CPU 1: cspot
[... 480 lines cut ...]
0x40086829: esp_crosscore_isr at C:/Users/EHaq/esp/esp-idf/components/esp32/crosscore_int.c:77

0x400844f9: _xt_lowint1 at C:/Users/EHaq/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x400fb172: char* std::__copy_move<false, false, std::random_access_iterator_tag>::__copy_m<unsigned char*, char*>(unsigned char*, unsigned char*, char*) at c:\espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\bits/stl_algobase.h:324
 (inlined by) char* std::__copy_move_a<false, unsigned char*, char*>(unsigned char*, unsigned char*, char*) at c:\espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\bits/stl_algobase.h:386
 (inlined by) char* std::__copy_move_a2<false, __gnu_cxx::__normal_iterator<unsigned char*, std::vector<unsigned char, std::allocator<unsigned char> > >, char*>(__gnu_cxx::__normal_iterator<unsigned char*, std::vector<unsigned char, std::allocator<unsigned char> > >, __gnu_cxx::__normal_iterator<unsigned char*, std::vector<unsigned char, std::allocator<unsigned char> > >, char*) at c:\espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\bits/stl_algobase.h:422
 (inlined by) char* std::copy<__gnu_cxx::__normal_iterator<unsigned char*, std::vector<unsigned char, std::allocator<unsigned
char> > >, char*>(__gnu_cxx::__normal_iterator<unsigned char*, std::vector<unsigned char, std::allocator<unsigned char> > >, __gnu_cxx::__normal_iterator<unsigned char*, std::vector<unsigned char, std::allocator<unsigned char> > >, char*) at c:\espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\bits/stl_algobase.h:455
 (inlined by) vorbisReadCb at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/src/ChunkedAudioStream.cpp:7

0x400fb23a: _get_data at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/tremor/vorbisfile.c:73
 (inlined by) _get_next_page at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/tremor/vorbisfile.c:125

0x400fb93a: _fetch_and_process_packet at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/tremor/vorbisfile.c:614

0x400fcb72: ov_read at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/tremor/vorbisfile.c:1581

0x400faaab: ChunkedAudioStream::startPlaybackLoop() at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/src/ChunkedAudioStream.cpp:105

0x400f71d3: Player::runTask() at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/src/Player.cpp:79

0x40199beb: Task::taskEntryFunc(void*) at C:/Users/EHaq/Desktop/brian_stover/cspot/cspot/include/Task.h:41

0x400d2e80: pthread_task_func at C:/Users/EHaq/esp/esp-idf/components/pthread/pthread.c:209 (discriminator 15)

feelfreelinux, 2021-12-22:
Hi, 
Please check newest master, I have resolved a couple of issues

ehabulhaq, 2022-01-03:
Hi,

Thanks for your response. Yes the newest resolved the issue

closed 2022-01-03 by ehabulhaq