feelfreelinux/cspot #108: No AC101 detected (i2c driver not installed?)

CaseyNolte22, 2022-02-03:
Hey,

After spending the time getting this to compile and flash, I'm getting a "No AC101 detected" message in the serial monitor. (I also see "i2c driver not installed" but if my googling is correct this is a non-issue.)

I'm using [this ](https://www.aliexpress.com/item/1005002392615155.html?spm=a2g0o.9042311.0.0.4d444c4dLiRTmb)board which should have a AC101 according to your docs, I've spent awhile trying to confirm but I can not find the relevant documentation.

Does anyone have any guidance on a solution? Serial output below.

`I (5250) example_connect: Connected to example_connect: sta
I (5260) example_connect: - IPv4 address: 192.168.1.234
I (5260) example_connect: - IPv6 address: fe80:0000:0000:0000:a6e5:7cff:fee8:af78, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (5270) cspot: Connected to AP, start spotify receiver
W (5280) AC101: No AC101 detected
E (5280) i2c: i2c_master_cmd_begin(1394): i2c driver not installed
E (5290) i2c: i2c_master_cmd_begin(1394): i2c driver not installed
E (5300) i2c: i2c_master_cmd_begin(1394): i2c driver not installed
E (5300) i2c: i2c_master_cmd_begin(1394): i2c driver not installed

...
...

D MercuryManager.cpp:213: Received packet with code 178 of length 65
D AudioChunkManager.cpp:96: ID: 2: finalize chunk!
D AudioChunkManager.cpp:19: Chunk requested 3
D ChunkedAudioStream.cpp:87: --- Loaded file
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40106f0e  PS      : 0x00060c30  A0      : 0x800f67d0  A1      : 0x3ffde9a0
0x40106f0e: i2s_write at C:/Espressif/frameworks/esp-idf-v4.4/components/driver/i2s.c:2006 (discriminator 2)

A2      : 0x00000000  A3      : 0x3f8127d8  A4      : 0x000001c8  A5      : 0x3ffde9e0
A6      : 0xffffffff  A7      : 0x3ffc43e8  A8      : 0x00000000  A9      : 0x3ffde950
A10     : 0x00000001  A11     : 0x3ffde9b4  A12     : 0x00000000  A13     : 0x3ffb357c
A14     : 0x3ffb3528  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000018  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x40106f0b:0x3ffde9a00x400f67cd:0x3ffde9e0
0x40106f0b: i2s_write at C:/Espressif/frameworks/esp-idf-v4.4/components/driver/i2s.c:2006 (discriminator 2)

0x400f67cd: i2sFeed(void*) at C:/Espressif/frameworks/esp-idf-v4.4/projects/cspot/cspot/bell/src/sinks/esp/BufferedAudioSink.cpp:20

ELF file SHA256: 8fd06f88f5e348ec

Rebooting...`

HeikoGr, 2022-02-03:
The newer ai thinker boards are having ES8388 as DAC chips. What board version do you have (printed on the front) and what ESP32-A2S Version do you have? Printed on the back (of esp32, under the printed antenna)

CaseyNolte22, 2022-02-03:
The version is "V2.2 A149" and the only text printed on the back is "2974". I'll try switching the config to ES8388 tonight.

HeikoGr, 2022-02-03:
Can confirm - 2974 has ES8388 inside. You may need an older cspot version - i have the same promblems while compiling.

CaseyNolte22, 2022-02-03:
I apologize I'm not well versed enough on using git/github or contributing to projects to do a pull request, but I got the current cspot version to work with the ES8388 with a couple simple additions. It works decent, except you must turn the Spotify volume extremely low or else there is insane clipping. I'm not knowledgeable enough to find a fix for that. 

It looks feelfreelinux deleted the ES8388 code necessary in .../targets/esp32/main/main.cpp in: 027a52fa7fc7491f638c7cde2c50dbd643a4fae5. 

Here is the main.cpp I added to get it to work. Again sorry I can't do it the proper way.
[main.zip](https://github.com/feelfreelinux/cspot/files/7998111/main.zip)

HeikoGr, 2022-02-04:
i think @anabolyc solved the issue too in his PR #109 , so i won't make another PR out of your edit.

can you tell me if you used lineout or the headphone jack? Because i have no audio on lineout and wasn't able to test the headphone jack yesterday

CaseyNolte22, 2022-02-04:
Headphone jack, I did not test lineout, and don't have the right cables to test it.