feelfreelinux/cspot #112: Stutters when fetching chunk on esp32

TShapinsky, 2022-02-24:
every time `Chunk requested %d` prints on the monitor there is a tiny audio stutter. Is it possible that chunks aren't being pre-fetched enough? Is too much CPU being used fetching that it messes with the I2S output?

Setup:
Board: ESP32-WROVER-E (8MB PSRAM)
Codec: PCM5102A

KennyStier, 2022-05-12:
Having similar issues
Board: ESP32-A1S
Codec: ES8388

jdthorpe, 2022-10-14:
I wonder if by "stutter" you mean pops / clicks.  In my case, the pops were absolutely intolerable (20-30 clicks per minute) and sounded like an old record

After doing some reading on I2S Click problems, I found two solutions mentioned around the internets: (A) adding inline resistors to attenuate the ringing in the I2S traces (or breadboard connectors, in my case) and (B) reducing the slew rate by reducing the output capacity on the GPIO pins.   I tried the latter first (b/c I didn't have any room on my breadboard...) by adding the following lines in the appropriate sink file (`cspot/bell/src/sinks/...`) and the issue *almost* completely went away (1 click ever 2-3 minutes).

```cpp
    BELL_LOG(info, "i2s", "Setting pin drive capacity");
    ESP_ERROR_CHECK(
        gpio_set_drive_capability((gpio_num_t)mck, GPIO_DRIVE_CAP_0));
    ESP_ERROR_CHECK(
        gpio_set_drive_capability((gpio_num_t)bck, GPIO_DRIVE_CAP_0));
    ESP_ERROR_CHECK(
        gpio_set_drive_capability((gpio_num_t)ws, GPIO_DRIVE_CAP_0));
    ESP_ERROR_CHECK(
        gpio_set_drive_capability((gpio_num_t)dataOut, GPIO_DRIVE_CAP_0));
```

I think I'll try the inline resistors at some point, but for now this made a big difference in the audio quality for me...