feelfreelinux/cspot #126: trackinfo.restriction.size()=1, esp32 doesn't recieve any playable data

SirCooper, 2022-07-23:
I've been triying to build cspot with a vs1053, and it won't work. I think the main problem is, that the chunks, the player recieves, always stay empty. 
I've made a build with a non-existent ac101, just to show you the LOG without any extra messages.
I'm using a Spotify-family-account
Approaches, i've tried yet:
- changing the WIFI
- changing the Account to a non-Facebook-login-account
- printing pcmOut and any other callback, that's refering to the audioSink

`I ZeroconfAuthenticator.cpp:26: Got request for info
I ZeroconfAuthenticator.cpp:37: Got request for adding user
I main.cpp:125: Creating player
I (20838) statusled: Chaing status to 3
D Session.cpp:39: Connecting with AP <ap-gew4.spotify.com:4070>
D PlainConnection.cpp:70: Connected to spotify server
D Session.cpp:42: Sending APHello packet...
D Session.cpp:102: Processing AP hello response...
D Session.cpp:104: Received AP hello response
D Session.cpp:150: Received shannon keys
D Session.cpp:81: Authorization successful
I (21248) statusled: Chaing status to 4
E (21258) i2c: i2c_master_cmd_begin(1398): i2c driver not installed
E (21258) i2c: i2c_master_cmd_begin(1398): i2c driver not installed
D MercuryManager.cpp:299: executing MercuryType SUB
D MercuryManager.cpp:216: Received packet with code 2 of length 336
D MercuryManager.cpp:216: Received packet with code 118 of length 2
D MercuryManager.cpp:216: Received packet with code 27 of length 2
D MercuryManager.cpp:223: Received country code: CH
D MercuryManager.cpp:216: Received packet with code 80 of length 6287
D MercuryManager.cpp:216: Received packet with code 31 of length 17
D MercuryManager.cpp:216: Received packet with code 105 of length 0
D MercuryManager.cpp:216: Received packet with code 181 of length 513
D MercuryManager.cpp:216: Received packet with code 74 of length 4
D MercuryManager.cpp:216: Received packet with code 179 of length 50
D MercuryManager.cpp:299: executing MercuryType SEND
D SpircController.cpp:32: Sent kMessageTypeHello!
D MercuryManager.cpp:216: Received packet with code 178 of length 50
D MercuryManager.cpp:216: Received packet with code 181 of length 551
D SpircController.cpp:114: Notify frame
D MercuryManager.cpp:216: Received packet with code 181 of length 981
D SpircController.cpp:114: Notify frame
D MercuryManager.cpp:216: Received packet with code 181 of length 568
D SpircController.cpp:114: Notify frame
D MercuryManager.cpp:216: Received packet with code 181 of length 1024
D SpircController.cpp:149: Load frame!
I PlayerState.cpp:144: ---- Track count 16
I PlayerState.cpp:145: ---- Inner track count 0
D MercuryManager.cpp:299: executing MercuryType GET
D MercuryManager.cpp:299: executing MercuryType SEND
D MercuryManager.cpp:216: Received packet with code 178 of length 1105
D MercuryManager.cpp:254:  MercuryType::UNSUB response->parts[0].size() = 874
I SpotifyTrack.cpp:99: Track name: Bulibasha
I SpotifyTrack.cpp:100: Track duration: 205240
D SpotifyTrack.cpp:101: trackInfo.restriction.size() = 1
I main.cpp:149: Track Info
D MercuryManager.cpp:216: Received packet with code 178 of length 50
D MercuryManager.cpp:216: Received packet with code 13 of length 20
I SpotifyTrack.cpp:207: Successfully got audio key!
D AudioChunkManager.cpp:19: Chunk requested 0
D MercuryManager.cpp:44: Reconnection required, no mercury response
E PlainConnection.cpp:121: Connection lost, will need to reconnect...
D MercuryManager.cpp:129: Trying to reconnect...
I PlainConnection.cpp:175: Closing socket...
D Session.cpp:39: Connecting with AP <ap-gew4.spotify.com:4070>
D PlainConnection.cpp:70: Connected to spotify server
D Session.cpp:42: Sending APHello packet...
D Session.cpp:102: Processing AP hello response...
D Session.cpp:104: Received AP hello response
D Session.cpp:150: Received shannon keys
D Session.cpp:81: Authorization successful
D MercuryManager.cpp:147: Reconnected successfuly :)
D AudioChunkManager.cpp:19: Chunk requested 1
D MercuryManager.cpp:299: executing MercuryType SUB
D AudioChunkManager.cpp:19: Chunk requested 2
I main.cpp:152: Track Pause
D SpircController.cpp:50: External play command
D MercuryManager.cpp:299: executing MercuryType SEND
D MercuryManager.cpp:216: Received packet with code 10 of length 4
E MercuryManager.cpp:242: Audio Chunk failure!
D MercuryManager.cpp:216: Received packet with code 181 of length 549
I ChunkedByteStream.cpp:82: Chunk not loaded, waiting for 0
D SpircController.cpp:114: Notify frame
D MercuryManager.cpp:216: Received packet with code 2 of length 336
D MercuryManager.cpp:216: Received packet with code 118 of length 2
D MercuryManager.cpp:216: Received packet with code 27 of length 2
D MercuryManager.cpp:223: Received country code: CH
D MercuryManager.cpp:216: Received packet with code 80 of length 6287
D MercuryManager.cpp:216: Received packet with code 31 of length 17
D MercuryManager.cpp:216: Received packet with code 105 of length 0
D MercuryManager.cpp:216: Received packet with code 181 of length 513
D MercuryManager.cpp:216: Received packet with code 10 of length 4
E MercuryManager.cpp:242: Audio Chunk failure!
D MercuryManager.cpp:216: Received packet with code 179 of length 50
D MercuryManager.cpp:299: executing MercuryType SEND
D SpircController.cpp:32: Sent kMessageTypeHello!
D MercuryManager.cpp:216: Received packet with code 10 of length 4
E MercuryManager.cpp:242: Audio Chunk failure!
D MercuryManager.cpp:216: Received packet with code 74 of length 4
I D MercuryManager.cppChunkedByteStream.cpp:216: Received packet with code 178 of length 50
:74: Chunk not found, requesting 0
D AudioChunkManager.cpp:19: Chunk requested 3
I ChunkedByteStream.cpp:82: Chunk not loaded, waiting for 0
D MercuryManager.cpp:216: Received packet with code 178 of length 50
D MercuryManager.cpp:216: Received packet with code 10 of length 4
E MercuryManager.cpp:242: Audio Chunk failure!
D MercuryManager.cpp:216: Received packet with code 181 of length 551
I ChunkedByteStream.cpp:74: Chunk not found, requesting 0
D SpircController.cpp:114: Notify frame
D D MercuryManager.cppAudioChunkManager.cpp:216: Received packet with code 181 of length 549
:19: Chunk requested 4
D SpircController.cppI :114: Notify frame
ChunkedByteStream.cppD :82: Chunk not loaded, waiting for 0MercuryManager.cpp
:216: Received packet with code 181 of length 568
D SpircController.cpp:114: Notify frame
D MercuryManager.cpp:216: Received packet with code 10 of length 4
E MercuryManager.cpp:242: Audio Chunk failure!
I ChunkedByteStream.cpp:74: Chunk not found, requesting 0
D AudioChunkManager.cpp:19: Chunk requested 5
I ChunkedByteStream.cpp:82: Chunk not loaded, waiting for 0
D MercuryManager.cpp:216: Received packet with code 10 of length 4
E MercuryManager.cpp:242: Audio Chunk failure!`

HeikoGr, 2022-07-29:
take a look here: https://github.com/feelfreelinux/cspot/issues/127#issuecomment-1199164322
maybe it's the same but different.

markuspi, 2022-07-30:
After solving #127 I ran into this exact issue. Changing the AP address like @HeikoGr described in https://github.com/feelfreelinux/cspot/issues/127#issuecomment-1199164322 solved this issue for me

feelfreelinux, 2022-12-18:
Closed with new APResolve support

closed 2022-12-18 by feelfreelinux