feelfreelinux/cspot #132: Crash when receiving new track

philippe44, 2022-09-12:
Hi - I have a recurring crash always at the same place, and with different tracks (randomly happens). It happens with the squeezeesp32 full version but also with my standalone esp32 cspot test application. Here is a coredump. It seems that something is wrong with TrackInfo data. The protobuf are fresh-generated at every build. 

NB: In the log below, the track title is in Chinese but that happens with any track)

```
D MercuryManager.cpp:299: executing MercuryType GET
D MercuryManager.cpp:299: executing MercuryType SEND
D MercuryManager.cpp:216: Received packet with code 178 of length 941
D MercuryManager.cpp:254:  MercuryType::UNSUB response->parts[0].size() = 710
I SpotifyTrack.cpp:99: Track name: 我愛過
I SpotifyTrack.cpp:100: Track duration: 297866
D SpotifyTrack.cpp:101: trackInfo.restriction.size() = 1
I SpotifyTrack.cpp:107: Trying alternative 0
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400eb7ec  PS      : 0x00060e30  A0      : 0x800ec11e  A1      : 0x3ffdca20
0x400eb7ec: SpotifyTrack::canPlayTrack(int) at D:/Developments/msys2/home/esp32/cspot/cspot/src/SpotifyTrack.cpp:74

A2      : 0x3ffe2a24  A3      : 0x00000000  A4      : 0x0000006b  A5      : 0x3ffdca40
A6      : 0x3f409fbf  A7      : 0x3f40a682  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x3ffe2a24  A11     : 0x3ffdd354  A12     : 0x3ffdca00  A13     : 0x00000014
A14     : 0x0000ffff  A15     : 0x3ffdca40  SAR     : 0x00000007  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000034  LBEG    : 0x4008a7a2  LEND    : 0x4008a7ad  LCOUNT  : 0x00000000
0x4008a7a2: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:150

0x4008a7ad: memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:160

Backtrace:0x400eb7e9:0x3ffdca20 0x400ec11b:0x3ffdca40 0x400ec3fe:0x3ffdcb10 0x400e1743:0x3ffdcb40 0x400e3335:0x3ffdcb60 0x400e3369:0x3ffdcbd0 0x400da6c1:0x3ffdcc00 0x400e79d3:0x3ffdccb0 0x400e7c1b:0x3ffdcd80 0x400e7cc5:0x3ffdce40 0x400f0e1e:0x3ffdce60 0x400f1271:0x3ffdd050 0x400f1929:0x3ffdd0d0 0x400dab61:0x3ffdd210
0x400eb7e9: SpotifyTrack::canPlayTrack(int) at D:/Developments/msys2/home/esp32/cspot/cspot/src/SpotifyTrack.cpp:74

0x400ec11b: SpotifyTrack::trackInformationCallback(std::unique_ptr<MercuryResponse, std::default_delete<MercuryResponse> >, unsigned int, bool) at D:/Developments/msys2/home/esp32/cspot/cspot/src/SpotifyTrack.cpp:104

0x400ec3fe: std::_Function_handler<void (std::unique_ptr<MercuryResponse, std::default_delete<MercuryResponse> >), SpotifyTrack::SpotifyTrack(std::shared_ptr<MercuryManager>, std::shared_ptr<TrackReference>, unsigned int, bool)::{lambda(std::unique_ptr<MercuryResponse, std::default_delete<MercuryResponse> >)#1}>::_M_invoke(std::_Any_data const&, std::unique_ptr<MercuryResponse, std::default_delete<MercuryResponse> >&&) at D:/Developments/msys2/home/esp32/cspot/cspot/src/SpotifyTrack.cpp:17
 (inlined by) _M_invoke at d:\users\phili\.espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297

0x400e1743: std::function<void (std::unique_ptr<MercuryResponse, std::default_delete<MercuryResponse> >)>::operator()(std::unique_ptr<MercuryResponse, std::default_delete<MercuryResponse> >) const at d:\users\phili\.espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687

0x400e3335: MercuryManager::updateQueue() at D:/Developments/msys2/home/esp32/cspot/cspot/src/MercuryManager.cpp:259

0x400e3369: MercuryManager::handleQueue() at D:/Developments/msys2/home/esp32/cspot/cspot/src/MercuryManager.cpp:287
```

philippe44, 2022-09-12:
Silly question
```
   int altIndex = -1;
    while (!canPlayTrack(altIndex))
    {
        altIndex++;
        CSPOT_LOG(info, "Trying alternative %d", altIndex);
    
        if(altIndex > trackInfo.alternative_count) {
            // no alternatives for song
            return;
        }
    }
```
Shouldn'it be >= alternative_count? I'd assume count of 1 means index can be 0 only. I might not see the log because of the crash.

feelfreelinux, 2022-12-18:
Fixed :)

closed 2022-12-18 by feelfreelinux