feelfreelinux/cspot #134: addUserHandler does not work anymore

philippe44, 2022-11-16:


philippe44, 2022-11-16:
Got it - int the loop that fills the body with the HTTP request, it uses body.data() + read which does not increment the "size" attribute of body. So we end up with a std::string that has reserved space but an actual size of 0, so everything after that goes south. I will add a fix to my PR

latonita, 2022-11-22:
@philippe44 did you manage to fix this issue? After 4 hours of trying to select proper library versions I finally built it for esp32 and see same issue as you do.

philippe44, 2022-11-22:
Yes, see the PR I've pushed or my own repo

latonita, 2022-11-23:
@philippe44 apologies, i'm not sure i know how to find your PR :) can you please post a link to a fix- PR or place in your repo.

Basically, if i pull from your repo - the fix is there and i can try build it, right? :)

latonita, 2022-11-23:
I mean I see your change 
https://github.com/feelfreelinux/cspot/commit/7254726892f74d5acb998d8c8dd6d38cd0c072b3
and I use this version, however it still crashes, since request->body is empty

philippe44, 2022-11-23:
No it's here https://github.com/feelfreelinux/bell/blob/b0b770ed7f451f27f8eec8484fb14604c4fb9283/src/HTTPServer.cpp#L471

closed 2023-04-01 by philippe44