feelfreelinux/cspot #140: ESP32 CMake Errors

gthieleb, 2023-01-10:
I get some errors for esp32 build on recent master related to missing mdns package:

```
CMake Error at /home/user/esp-idf/tools/cmake/project.cmake:480 (add_executable):
  Target "cspot-esp32.elf" links to target "idf::mdns" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
Call Stack (most recent call first):
  CMakeLists.txt:13 (project)

CMake Error at /home/user/esp-idf/tools/cmake/component.cmake:484 (add_library):
  Target "__idf_main" links to target "idf::mdns" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
Call Stack (most recent call first):
  main/CMakeLists.txt:7 (idf_component_register)
```

My setup:

```
$ python --version
Python 3.10.6
which python
/home/user/.espressif/python_env/idf5.1_py3.10_env/bin/python
ls ~/mbedtls-v3.3.0/cmake/cmake/
MbedTLSConfig.cmake  MbedTLSConfigVersion.cmake  MbedTLSTargets.cmake
```

The linux build suceeded using this:

```
cd targets/cli
cmake .. -DBELL_EXTERNAL_MBEDTLS=/home/gun/mbedtls-v3.3.0/cmake/cmake -DMBEDTLS_RELEASE=NOCONFIG
```

HeikoGr, 2023-01-10:
I think it is because you use idf esp in version 5.1

You should try the 4.4 branch

gthieleb, 2023-01-11:
@HeikoGr  thanks, this is working better. I also tried with 4.4.3 yesterday but did not performed `git reset --hard`.

Currently I am struggling with the protobuf package. Installed is `protobuf  4.21.12` but this seem to not align with the protoc binary from my VM (Ubuntu 22.04).

gthieleb, 2023-01-11:
OK fixed protbuf so far. Another error on the linking step:

```
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/shared_ptr.h:359:59:   required from 'std::shared_ptr<_Tp>::shared_ptr(std::_Sp_alloc_shared_tag<_Tp>, _Args&& ...) [with _Alloc = std::allocator<ZeroconfAuthenticator>; _Args = {cspotTask(void*)::<lambda(std::shared_ptr<LoginBlob>)>&, std::shared_ptr<bell::HTTPServer>&}; _Tp = ZeroconfAuthenticator]'
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/shared_ptr.h:705:14:   required from 'std::shared_ptr<_Tp> std::allocate_shared(const _Alloc&, _Args&& ...) [with _Tp = ZeroconfAuthenticator; _Alloc = std::allocator<ZeroconfAuthenticator>; _Args = {cspotTask(void*)::<lambda(std::shared_ptr<LoginBlob>)>&, std::shared_ptr<bell::HTTPServer>&}]'
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/shared_ptr.h:721:39:   required from 'std::shared_ptr<_Tp> std::make_shared(_Args&& ...) [with _Tp = ZeroconfAuthenticator; _Args = {cspotTask(void*)::<lambda(std::shared_ptr<LoginBlob>)>&, std::shared_ptr<bell::HTTPServer>&}]'
/home/user/cspot/targets/esp32/main/main.cpp:195:98:   required from here
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/ext/new_allocator.h:136:4: error: no matching function for call to 'ZeroconfAuthenticator::ZeroconfAuthenticator(cspotTask(void*)::<lambda(std::shared_ptr<LoginBlob>)>&, std::shared_ptr<bell::HTTPServer>&)'
  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/user/cspot/targets/esp32/main/main.cpp:24:
/home/user/cspot/cspot/include/ZeroconfAuthenticator.h:53:5: note: candidate: 'ZeroconfAuthenticator::ZeroconfAuthenticator(authCallback, std::shared_ptr<bell::BaseHTTPServer>, std::__cxx11::string, std::__cxx11::string, void*)'
     ZeroconfAuthenticator(authCallback callback, std::shared_ptr<bell::BaseHTTPServer> httpServer, std::string name, std::string deviceId, void *mdnsService = NULL);
     ^~~~~~~~~~~~~~~~~~~~~
/home/user/cspot/cspot/include/ZeroconfAuthenticator.h:53:5: note:   candidate expects 5 arguments, 2 provided
/home/user/cspot/cspot/include/ZeroconfAuthenticator.h:36:7: note: candidate: 'ZeroconfAuthenticator::ZeroconfAuthenticator(ZeroconfAuthenticator&&)'
 class ZeroconfAuthenticator {
       ^~~~~~~~~~~~~~~~~~~~~
/home/user/cspot/cspot/include/ZeroconfAuthenticator.h:36:7: note:   candidate expects 1 argument, 2 provided
In file included from /home/user/cspot/cspot/include/LoginBlob.h:7,
                 from /home/user/cspot/cspot/include/Session.h:13,
                 from /home/user/cspot/targets/esp32/main/main.cpp:21:
``´

HeikoGr, 2023-01-12:
i have no solution but you may try to
`git submodule update --init --recursive`

and remove/empty the 'build' directory below 'cspot/targets/esp32/', maybe there are some artefacts from esp-idf 5.1

gthieleb, 2023-01-12:
I did remove tge whole `~/.espressif` folder. Then i reinstalled completely. The `build` was deleted after each failed build. 

Is there a possibility to show the version of the `ZeroconfAuthenticator` component? 

Do you know if this is shipped together with IDF or a submodule in the repo?

HeikoGr, 2023-01-12:
As far as i know it’s part of libavahi-compat-libdnssd.

but unfortunately i get the same error. The last checkout i can successfully compile is 104d70a

@feelfreelinux : can you help us?

gthieleb, 2023-02-22:
Late update, but with the mentioned commit i was able to build.

MainManu, 2023-10-16:
I am having the exact same issues. I cannot for the life of me build. I tried so far: 
- uninstalling idf > 5.x
- installing idf 4.4
- deleting .esp folder and running the idf install script
- downgrading protobuf to 3.20
so far nothing worked.