feelfreelinux/cspot #144: Question

itsjustyryr, 2023-03-09:
Can it play music in the background?

itsjustyryr, 2023-03-09:
That's what I've done, waiting for an answer! Fingers crossed, and thanks!

Ryan-Romig, 2023-03-09:
it looks like to me  Cspot-vita uses the Cspot library as a submodule to work on PS vita platform instead of ESP32 or CLI. not the whole Cspot project. 

similar to adding a PS-Vita folder to the cspot/targets if that makes sense

itsjustyryr, 2023-03-09:
> it looks like to me  Cspot-vita uses the Cspot library as a submodule to work on PS vita platform instead of ESP32 or CLI. not the whole Cspot project. 
> 
> 
> 
> similar to adding a PS-Vita folder to the cspot/targets if that makes sense

I also noticed that! I was curious if one of them has the ability to play music in the background, even if you play in Adrenaline(emulator for PSP/PS1 games)

Ryan-Romig, 2023-03-09:
> I also noticed that! I was curious if one of them has the ability to play music in the background, even if you play in Adrenaline(emulator for PSP/PS1 games)

what do you mean in the background? do you mean on the PS-Vita platform?
If you are running an emulator on a PC, you could run CSpot CLI and the emulator both.

itsjustyryr, 2023-03-10:
> what do you mean in the background? do you mean on the PS-Vita platform?
>
> If you are running an emulator on a PC, you could run CSpot CLI and the emulator both.
Sorry. On the PS-Vita platform, yes. Can it be used with LowMemMode so the app would run in the background?

closed 2023-04-03 by itsjustyryr