feelfreelinux/cspot #153: CSPOT device not listed on Spotify Web API

marcusathalide, 2023-05-17:
Hello all,

I got CSPOT running on a ESP32. It is possible to connect via Spotify App from my phone and play music.

However, using the Spotify Web API (Get Available Devices), the CSPOT devices does not show up. (I can use the Web API to fetch my playlists - so overall auth did work).

It seems that "thrid-party" devices are not supported via the Web API. 

Did anyone here succeed in controlling the CSPOT via Spotify Web API?

Many thanks in advance
Marcus

sorryusernameisalreadytaken, 2023-07-26:
I had the same problem. But I run a [Squeezelite-ESP32 (Generic/I2S) ](https://sle118.github.io/squeezelite-esp32-installer/).
I think it is a cspot error, so I land here.

philippe44, 2023-09-20:
Same here, I realized that when integrating with HomeAssistant

philippe44, 2023-09-25:
Well, just thought about it more 🤓. ZeroConf is used by default but that means that Spotify servers have no way to see local players, as they are not registered (not discovered) yet. They will only be when a local applicatio on your network discovers them and then registers them.

You need to use login/password so that the player is known and then you'll see them from the webAPI

philippe44, 2023-09-25:
Having said that, unfortunately, there is still a problem where the authorization works as expected but getting the AccessKey fails all the time

philippe44, 2023-09-25:
Right, my understanding is that now that Spotify's CDN is used, we need credentials to get an access token/key (was not the case before). These credentials are given by Spotify controller when using ZeroConf but when using username/password, they are return as part of the authentication process. These were discarded previously but now they probably need to be used. I'll see if I can figure out something or hopefully @feelfreelinux will jump in and do in 5 min what will take me 5 days 😄.

philippe44, 2023-09-25:
Done - it took me 5 hours which is still 295 minutes more than @feelfreelinux. I'll submit a PR soon

philippe44, 2023-09-26:
It's fixed now, you just need to sue username/password once. After that, you can create a json credential files to be re-used for next sessions.

marcusathalide, 2023-12-30:
Hello @philippe44,
First of all, thanks for looking into this. 
I just had some time to try out the committed changes.
Unfortunately, for me it does not yet work. I tried the following:
1) Built the cspotcli 
2) Started cspotcli without any parameters
Result: Cspot is visiable in the SpotiyApp but not in the Spotify Web API (as expected)

3) Started cspot with -u <spotifyusername> and -p <spotifypassword> parameters 
[2023-12-30 17:21:20.573] I main.cpp:195: Creating player
[2023-12-30 17:21:20.642] D Session.cpp:67: Connecting with AP <ap-gew4.spotify.com:4070>
[2023-12-30 17:21:20.660] D PlainConnection.cpp:101: Connected to spotify server
[2023-12-30 17:21:20.684] I Session.cpp:43: Received APHello response
[2023-12-30 17:21:20.688] D Session.cpp:48: Received shannon keys
**[2023-12-30 17:21:20.788] D Session.cpp:87: Authorization successful**
[2023-12-30 17:21:20.789] I MercurySession.cpp:42: Received packet, command: 4
[2023-12-30 17:21:20.789] D TimeProvider.cpp:15: Time synced with spotify servers
[2023-12-30 17:21:20.789] I MercurySession.cpp:42: Received packet, command: 2
[2023-12-30 17:21:20.789] I MercurySession.cpp:42: Received packet, command: 118
[2023-12-30 17:21:20.789] I MercurySession.cpp:42: Received packet, command: 27
[2023-12-30 17:21:20.789] I MercurySession.cpp:42: Received packet, command: 80
[2023-12-30 17:21:20.789] I MercurySession.cpp:42: Received packet, command: 31
[2023-12-30 17:21:20.789] I MercurySession.cpp:42: Received packet, command: 105
[2023-12-30 17:21:20.805] I MercurySession.cpp:42: Received packet, command: 181
[2023-12-30 17:21:20.864] I MercurySession.cpp:42: Received packet, command: 74
[2023-12-30 17:21:20.874] D MercurySession.cpp:251: Executing Mercury Request, type SUB
[2023-12-30 17:21:20.874] D MercurySession.cpp:153: Received country code DE
[2023-12-30 17:21:20.889] I AccessKeyFetcher.cpp:99: Access token expired, fetching new one... 241
[2023-12-30 17:21:20.893] I MercurySession.cpp:42: Received packet, command: 179
[2023-12-30 17:21:20.893] D MercurySession.cpp:174: Received mercury packet
[2023-12-30 17:21:20.893] D MercurySession.cpp:251: Executing Mercury Request, type SEND
[2023-12-30 17:21:20.893] D SpircHandler.cpp:61: Sent kMessageTypeHello!
[2023-12-30 17:21:20.912] I MercurySession.cpp:42: Received packet, command: 178
[2023-12-30 17:21:20.912] D MercurySession.cpp:174: Received mercury packet
[2023-12-30 17:21:20.573] I main.cpp:195: Creating player
[2023-12-30 17:21:20.642] D Session.cpp:67: Connecting with AP <ap-gew4.spotify.com:4070>
[2023-12-30 17:21:20.660] D PlainConnection.cpp:101: Connected to spotify server
[2023-12-30 17:21:20.684] I Session.cpp:43: Received APHello response
[2023-12-30 17:21:20.688] D Session.cpp:48: Received shannon keys
[2023-12-30 17:21:20.788] D Session.cpp:87: Authorization successful
[2023-12-30 17:21:20.789] I MercurySession.cpp:42: Received packet, command: 4
[2023-12-30 17:21:20.789] D TimeProvider.cpp:15: Time synced with spotify servers
[2023-12-30 17:21:20.789] I MercurySession.cpp:42: Received packet, command: 2
[2023-12-30 17:21:20.789] I MercurySession.cpp:42: Received packet, command: 118
[2023-12-30 17:21:20.789] I MercurySession.cpp:42: Received packet, command: 27
[2023-12-30 17:21:20.789] I MercurySession.cpp:42: Received packet, command: 80
[2023-12-30 17:21:20.789] I MercurySession.cpp:42: Received packet, command: 31
[2023-12-30 17:21:20.789] I MercurySession.cpp:42: Received packet, command: 105
[2023-12-30 17:21:20.805] I MercurySession.cpp:42: Received packet, command: 181
[2023-12-30 17:21:20.864] I MercurySession.cpp:42: Received packet, command: 74
[2023-12-30 17:21:20.874] D MercurySession.cpp:251: Executing Mercury Request, type SUB
[2023-12-30 17:21:20.874] D MercurySession.cpp:153: Received country code DE
[2023-12-30 17:21:20.889] I AccessKeyFetcher.cpp:99: Access token expired, fetching new one... 241
[2023-12-30 17:21:20.893] I MercurySession.cpp:42: Received packet, command: 179
[2023-12-30 17:21:20.893] D MercurySession.cpp:174: Received mercury packet
[2023-12-30 17:21:20.893] D MercurySession.cpp:251: Executing Mercury Request, type SEND
[2023-12-30 17:21:20.893] D SpircHandler.cpp:61: Sent kMessageTypeHello!
[2023-12-30 17:21:20.912] I MercurySession.cpp:42: Received packet, command: 178
[2023-12-30 17:21:20.912] D MercurySession.cpp:174: Received mercury packet
**[2023-12-30 17:21:20.976] I AccessKeyFetcher.cpp:114: Access token sucessfully fetched**

So it looks like access tokens can be fetched. 
Unfortunately, starting cspotcli in this way, it is not visable in neither the SpotiyApp nor the Spotify Web API.

Any idea, what I could have done wrong here?

Many thanks again in advance for the help
Marcus

philippe44, 2023-12-30:
This is weird, I'm using it now with the option to store credentials all the time. I'll verify with -u -p

marcusathalide, 2023-12-30:
After starting the SpotifyApp once, it got visible also in the Web API... Sorry I should have read the instructions probably.

Sorry!

marcusathalide, 2023-12-30:
But it seems then that the -u/-p/-c options are not required.
I was hoping to use the cspot via a Web API without using the SpotifyApps at all. Should this be possible with the -u/-p/-c options?

philippe44, 2023-12-30:
Nono don't think it should. Try the -c and create/store credentials once. Then restart it

marcusathalide, 2023-12-30:
I did the following now:
1) Close SpotifyApp 
2) Restart cspotcli -c test.json (on the console is see: AccessKeyFetcher.cpp:114: Access token sucessfully fetched) - so it seems working.
3) Open the Spotify WebPlayer API - the CSPOT device is not showing up under "Connect to a device".
4) Starting the SpotifyApp - also here the device is not "detected" (if I start the cspotcli without options it shows up here).

philippe44, 2023-12-30:
Here is the trick that works for me: you have 2 identifiers on Spotify: your email address and a profile name. You can use both with the same password on -u and -p, but only if you use the profile name will the player appear in the webUI . Don't ask me exactly why but I think credential should also be created the same way, using <profile> for -U, not the email. 

Remember as well that the .json file can be created without using -u and -p, just use -c and connect **once** using the desktop app, it will create the credential files.

marcusathalide, 2023-12-31:
Hello @philippe44,

that is it, I use the email as username before and not the profile name - now it works!

Thanks a lot for the hints and support.

Marcus

felipebaisi, 2024-02-08:
YES!!!!! @philippe44 
It works! 
[image]