feelfreelinux/cspot #156: ESP-IDF version for ESP32 target

myreauks, 2023-09-15:
Hi!
I've been trying to build the master branch with IDF 4.4 and 5.0 but both fails with varying error messages. Before starting a more in-depth research on the error messages, is there a recommendation on which version of the IDF to use?

Furthermore, has anyone succesfully compiled the master branch on ESP32?

myreauks, 2023-09-15:
A little update on the situation.
I have gotten pretty far now by installing the v5.1.1 version of ESP-IDF and manually installing the led_strip and mdns components.
I have so far managed to succesfully set the target to ESP32 and setup it with the menuconfig.
With the build I am getting to step 1222/1500 so almost getting there now.

However, at that point, I am starting to get these errors. Getting a bit overwhelmed at this point. Any pointers?

```
[1222/1500] Building CXX object esp-idf/main/cspot/bell/CMakeFiles/bell.dir/main/io/BinaryStream.cpp.obj
FAILED: esp-idf/main/cspot/bell/CMakeFiles/bell.dir/main/io/BinaryStream.cpp.obj 
/Users/user/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM -DFMT_HEADER_ONLY -DIDF_VER=\"v5.1.1\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DPB_ENABLE_MALLOC -DPB_FIELD_32BIT -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DUSE_DEFAULT_STDLIB="1 -DTARGET_OS_IPHONE=0" -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/user/esp/cspot/targets/esp32/build/config -I/Users/user/esp/cspot/cspot/bell/main/audio-codec/include -I/Users/user/esp/cspot/cspot/bell/main/audio-dsp/include -I/Users/user/esp/cspot/cspot/bell/main/audio-sinks/include -I/Users/user/esp/cspot/cspot/bell/main/io/include -I/Users/user/esp/cspot/cspot/bell/main/utilities/include -I/Users/user/esp/cspot/cspot/bell/main/platform -I/Users/user/esp/cspot/cspot/bell/external/mqtt/include -I/Users/user/esp/cspot/cspot/bell/main/audio-containers/include -I/Users/user/esp/cspot/cspot/bell/main/audio-sinks/include/esp -I/Users/user/esp/cspot/cspot/bell/external/civetweb/include -I/Users/user/esp/cspot/targets/esp32/build/esp-idf/main/cspot/bell -I/Users/user/esp/esp-idf/components/newlib/platform_include -I/Users/user/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/user/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/user/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/Users/user/esp/esp-idf/components/freertos/esp_additions/include -I/Users/user/esp/esp-idf/components/freertos/esp_additions/arch/xtensa/include -I/Users/user/esp/esp-idf/components/esp_hw_support/include -I/Users/user/esp/esp-idf/components/esp_hw_support/include/soc -I/Users/user/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/Users/user/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/Users/user/esp/esp-idf/components/esp_hw_support/port/esp32/private_include -I/Users/user/esp/esp-idf/components/heap/include -I/Users/user/esp/esp-idf/components/log/include -I/Users/user/esp/esp-idf/components/soc/include -I/Users/user/esp/esp-idf/components/soc/esp32 -I/Users/user/esp/esp-idf/components/soc/esp32/include -I/Users/user/esp/esp-idf/components/hal/esp32/include -I/Users/user/esp/esp-idf/components/hal/include -I/Users/user/esp/esp-idf/components/hal/platform_port/include -I/Users/user/esp/esp-idf/components/esp_rom/include -I/Users/user/esp/esp-idf/components/esp_rom/include/esp32 -I/Users/user/esp/esp-idf/components/esp_rom/esp32 -I/Users/user/esp/esp-idf/components/esp_common/include -I/Users/user/esp/esp-idf/components/esp_system/include -I/Users/user/esp/esp-idf/components/esp_system/port/soc -I/Users/user/esp/esp-idf/components/esp_system/port/include/private -I/Users/user/esp/esp-idf/components/xtensa/include -I/Users/user/esp/esp-idf/components/xtensa/esp32/include -I/Users/user/esp/esp-idf/components/esp_timer/include -I/Users/user/esp/esp-idf/components/lwip/include -I/Users/user/esp/esp-idf/components/lwip/include/apps -I/Users/user/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/user/esp/esp-idf/components/lwip/lwip/src/include -I/Users/user/esp/esp-idf/components/lwip/port/include -I/Users/user/esp/esp-idf/components/lwip/port/freertos/include -I/Users/user/esp/esp-idf/components/lwip/port/esp32xx/include -I/Users/user/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/Users/user/esp/esp-idf/components/mdns/include -I/Users/user/esp/esp-idf/components/console -I/Users/user/esp/esp-idf/components/vfs/include -I/Users/user/esp/esp-idf/components/esp_netif/include -I/Users/user/esp/esp-idf/components/esp_event/include -I/Users/user/esp/esp-idf/components/mbedtls/port/include -I/Users/user/esp/esp-idf/components/mbedtls/mbedtls/include -I/Users/user/esp/esp-idf/components/mbedtls/mbedtls/library -I/Users/user/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/Users/user/esp/esp-idf/components/pthread/include -I/Users/user/esp/esp-idf/components/driver/include -I/Users/user/esp/esp-idf/components/driver/deprecated -I/Users/user/esp/esp-idf/components/driver/analog_comparator/include -I/Users/user/esp/esp-idf/components/driver/dac/include -I/Users/user/esp/esp-idf/components/driver/gpio/include -I/Users/user/esp/esp-idf/components/driver/gptimer/include -I/Users/user/esp/esp-idf/components/driver/i2c/include -I/Users/user/esp/esp-idf/components/driver/i2s/include -I/Users/user/esp/esp-idf/components/driver/ledc/include -I/Users/user/esp/esp-idf/components/driver/mcpwm/include -I/Users/user/esp/esp-idf/components/driver/parlio/include -I/Users/user/esp/esp-idf/components/driver/pcnt/include -I/Users/user/esp/esp-idf/components/driver/rmt/include -I/Users/user/esp/esp-idf/components/driver/sdio_slave/include -I/Users/user/esp/esp-idf/components/driver/sdmmc/include -I/Users/user/esp/esp-idf/components/driver/sigma_delta/include -I/Users/user/esp/esp-idf/components/driver/spi/includ [... cut ...]
/Users/user/esp/cspot/cspot/src/TrackQueue.cpp: In member function 'void cspot::QueuedTrack::stepParseMetadata(Track*, Episode*)':
/Users/user/esp/cspot/cspot/src/TrackQueue.cpp:151:7: warning: unused variable 'alternativeCount' [-Wunused-variable]
  151 |   int alternativeCount, filesCount = 0;
      |       ^~~~~~~~~~~~~~~~
/Users/user/esp/cspot/cspot/src/TrackQueue.cpp:152:8: warning: unused variable 'canPlay' [-Wunused-variable]
  152 |   bool canPlay = false;
      |        ^~~~~~~
ninja: build stopped: subcommand failed.
```

LaurentMe, 2023-09-17:
I'm having the exact same issue, if I make any progress I'll let you know!

myreauks, 2023-09-18:
I managed to fix the error with 'endian' not being recognised by modifying the #ifndef from the following lines in BinaryStream.h:

`#ifndef ESP_PLATFORM
#include <bit>  // for endian
#endif`

This makes me think that somehow the ESP_PLATFORM definition is not being passed here.
The rest of the errors are still there, but those might have to do with the same issue.

myreauks, 2023-10-13:
I have tried this again with the newest commits and the build still starts to fail in step 1226 with the TrackPlayer with the following error message:
```
[1225/1616] Building CXX object esp-idf/main/cspot/bell/CMakeFiles/bell.dir/main/io/BellMQTTClient.cpp.obj
/Users/user/esp/cspot/cspot/bell/main/io/BellMQTTClient.cpp: In member function 'void bell::MQTTClient::connect(const std::string&, uint16_t, const std::string&, const std::string&)':
/Users/user/esp/cspot/cspot/bell/main/io/BellMQTTClient.cpp:36:7: warning: unused variable 'status' [-Wunused-variable]
   36 |   int status = fcntl(socket.getFd(), F_SETFL,
      |       ^~~~~~
[1226/1616] Building CXX object esp-idf/main/cspot/CMakeFiles/cspot.dir/src/TrackPlayer.cpp.obj
/Users/user/esp/cspot/cspot/src/TrackPlayer.cpp: In member function 'virtual void cspot::TrackPlayer::runTask()':
/Users/user/esp/cspot/cspot/src/TrackPlayer.cpp:192:15: warning: unused variable 'r' [-Wunused-variable]
  192 |       int32_t r =
      |               ^
[1227/1616] Building CXX object esp-idf/main/cspot/bell/CMakeFiles/bell.dir/main/io/BufferedStream.cpp.obj
[1228/1616] Building CXX object esp-idf/main/cspot/bell/CMakeFiles/bell.dir/main/io/FileStream.cpp.obj
[1229/1616] Building CXX object esp-idf/main/cspot/bell/CMakeFiles/bell.dir/main/io/MGStreamAdapter.cpp.obj
[1230/1616] Building C object esp-idf/main/cspot/bell/CMakeFiles/bell.dir/main/io/picohttpparser.c.obj
[1231/1616] Building CXX object esp-idf/main/cspot/CMakeFiles/cspot.dir/src/MercurySession.cpp.obj
/Users/user/esp/cspot/cspot/src/MercurySession.cpp: In member function 'cspot::MercurySession::Response cspot::MercurySession::decodeResponse(const std::vector<unsigned char>&)':
/Users/user/esp/cspot/cspot/src/MercurySession.cpp:222:8: warning: unused variable 'sequenceLength' [-Wunused-variable]
  222 |   auto sequenceLength = ntohs(extract<uint16_t>(data, 0));
      |        ^~~~~~~~~~~~~~
/Users/user/esp/cspot/cspot/src/MercurySession.cpp:225:8: warning: unused variable 'partsNumber' [-Wunused-variable]
  225 |   auto partsNumber = ntohs(extract<uint16_t>(data, 11));
      |        ^~~~~~~~~~~
[1232/1616] Building CXX object esp-idf/main/cspot/bell/CMakeFiles/bell.dir/main/io/EncodedAudioStream.cpp.obj
[1233/1616] Building CXX object esp-idf/main/cspot/CMakeFiles/cspot.dir/src/PlaybackState.cpp.obj
[1234/1616] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj 
/Users/user/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM -DFMT_HEADER_ONLY -DIDF_VER=\"v5.1.1\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DPB_ENABLE_MALLOC -DPB_FIELD_32BIT -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DUNITY_INCLUDE_CONFIG_H -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/user/esp/cspot/targets/esp32/build/config -I/Users/user/esp/cspot/targets/esp32/main -I/Users/user/esp/esp-idf/components/newlib/platform_include -I/Users/user/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/user/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/user/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/Users/user/esp/esp-idf/components/freertos/esp_additions/include -I/Users/user/esp/esp-idf/components/freertos/esp_additions/arch/xtensa/include -I/Users/user/esp/esp-idf/components/esp_hw_support/include -I/Users/user/esp/esp-idf/components/esp_hw_support/include/soc -I/Users/user/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/Users/user/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/Users/user/esp/esp-idf/components/esp_hw_support/port/esp32/private_include -I/Users/user/esp/esp-idf/components/heap/include -I/Users/user/esp/esp-idf/components/log/include -I/Users/user/esp/esp-idf/components/soc/include -I/Users/user/esp/esp-idf/components/soc/esp32 -I/Users/user/esp/esp-idf/components/soc/esp32/include -I/Users/user/esp/esp-idf/components/hal/esp32/include -I/Users/user/esp/esp-idf/components/hal/include -I/Users/user/esp/esp-idf/components/hal/platform_port/include -I/Users/user/esp/esp-idf/components/esp_rom/include -I/Users/user/esp/esp-idf/components/esp_rom/include/esp32 -I/Users/user/esp/esp-idf/components/esp_rom/esp32 -I/Users/user/esp/esp-idf/components/esp_common/include -I/Users/user/esp/esp-idf/components/esp_system/include -I/Users/user/esp/esp-idf/components/esp_system/port/soc -I/Users/user/esp/esp-idf/components/esp_system/port/include/private -I/Users/user/esp/esp-idf/components/xtensa/include -I/Users/user/esp/esp-idf/components/xtensa/esp32/include -I/Users/user/esp/esp-idf/components/esp_timer/include -I/Users/user/esp/esp-idf/components/lwip/include -I/Users/user/esp/esp-idf/components/lwip/include/apps -I/Users/user/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/user/esp/esp-idf/components/lwip/lwip/src/include -I/Users/user/esp/esp-idf/components/lwip/port/include -I/Users/user/esp/esp-idf/components/lwip/port/freertos/include -I/Users/user/esp/esp-idf/components/lwip/port/esp32xx/include -I/Users/user/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/Users/user/esp/esp-idf/components/mdns/include -I/Users/user/esp/esp-idf/components/console -I/Users/user/esp/esp-idf/components/vfs/include -I/Users/user/esp/esp-idf/components/esp_netif/include -I/Users/user/esp/esp-idf/components/esp_event/include -I/Users/user/esp/cspot/cspot/include -I/Users/user/esp/cspot/targets/esp32/ [... cut ...]
/Users/user/esp/cspot/cspot/src/TrackQueue.cpp: In member function 'void cspot::QueuedTrack::stepParseMetadata(Track*, Episode*)':
/Users/user/esp/cspot/cspot/src/TrackQueue.cpp:151:7: warning: unused variable 'alternativeCount' [-Wunused-variable]
  151 |   int alternativeCount, filesCount = 0;
      |       ^~~~~~~~~~~~~~~~
/Users/user/esp/cspot/cspot/src/TrackQueue.cpp:152:8: warning: unused variable 'canPlay' [-Wunused-variable]
  152 |   bool canPlay = false;
      |        ^~~~~~~
ninja: build stopped: subcommand failed.

```

playduck, 2023-10-19:
Hi, @myreauks when you mention 
>  manually installing the led_strip and mdns components.
I'm wondering how exactly you are doing this considering the led_strip is an idf example project and afaik not meant to be imported?
I'm running into cmake errors when I try to manually move the led_strip example project within the idf.

myreauks, 2023-10-20:
> Hi, @myreauks when you mention

Hi,
I downloaded the library from here:
[https://components.espressif.com/components/espressif/led_strip](url)

After that I manually added the library folder to the following path: esp-idf/examples/common-components/
Not sure if this is the correct way to go about it, but at least for me it fixed the compile errors related to led-strip library.

tobiasguyer, 2023-12-20:
You can use my fork, should work for esp32, if you'll use a esp32 s3, you'll have to edit the sdkconfig file

playduck, 2023-12-21:
I've gotten this to build using @tobiasguyer's fork (I haven't test it with this official repo yet and I haven't tested it on HW. The build finishes without errors though).

I'm using esp-idf-v5.1.1.
Things I've modified in cspot:

- `targets/esp32/CMakeLists.txt`: Commented out line 8 -> `# list(APPEND EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/led_strip)`
- `targets/esp32/main/CMakeLists.txt`: Added after line 11 (inside `idf_component_register`) -> `    REQUIRES led_strip mdns spiffs esp_wifi nvs_flash protocol_examples_common
`

The components `spiffs`, `esp_wifi`, `nvs_flash` are from esp-idf.
You can get the modules `led_strip` and `mdns` from https://components.espressif.com by running:
```
$> idf.py add-dependency "espressif/mdns^1.2.2"
```

myreauks, 2023-12-21:
Hi!
Thanks for the tips!
I got it to build further than before with this, but it fails at 1331/1671 with the following error message:
```
In file included from /Users/myreauks/git/cspot/cspot/bell/main/io/include/BellHTTPServer.h:15,
                 from /Users/myreauks/git/cspot/cspot/bell/main/io/BellHTTPServer.cpp:1:
/Users/myreauks/git/cspot/cspot/bell/external/civetweb/include/CivetServer.h:291:22: error: 'virtual void CivetWebSocketHandler::handleClose(CivetServer*, const mg_connection*)' was hidden [-Werror=overloaded-virtual=]
  291 |         virtual void handleClose(CivetServer *server,
```

Are you using a certain commit of Bell or how did you manage to get past this?

playduck, 2023-12-21:
I didn't actually do anything with Bell. 
Maybe you're using an old/modified version of the idf?
Here's the list of steps I just used to reproduce a build:

```sh
mkdir temp
cd temp
# get idf version 5.1.1 and install it
git clone -b v5.1.1 --recursive https://github.com/espressif/esp-idf.git esp-idf-v5.1.1\n
cd esp-idf-v5.1.1
./install.sh
source ./export.sh
cd ..
# get the forked version of cspot (haven't tested the original version yet)
git clone --recursive https://github.com/tobiasguyer/cspot.git
cd cspot/targets/esp32
code . #make the aforementioned changes to the two cmake files in your favourite editor
# get the two required dependencies
idf.py add-dependency "espressif/led_strip^2.5.2"
idf.py add-dependency "espressif/mdns^1.2.2"
idf.py set-target esp32
idf.py menuconfig # no changes required here
idf.py build
# project build complete
```

myreauks, 2023-12-22:
This is great thanks! It seems that I had a newer version of the idf and that was the problem. Changing to 5.1.1 fixed everything! I will still have to verify the functionality on a board, but seems very promising.

xris99, 2023-12-23:
Thank you. Your patch build well, however on the board (esp32-wrover) I get a core dump directly after wifi connection and cspot starts. No messages between start and crash. 
I'm using a PCM5102 interface. Looks like a null-pointer to me.

tobiasguyer, 2023-12-24:
All good.. i'll update a working repo in the next days.. first problem is the defconf.default, wich doesn't include the psram settings, and zeroconf(the cspot task) needs to be called from a specific core.
I'll check my working repo and update it as soon as possible..

xris99, 2023-12-24:
Cool, thank you so much. Looking forward to it and happy to test.

tobiasguyer, 2023-12-25:
i've fixed it! let me know if it works for you

xris99, 2023-12-25:
Thank you so much.
Now I get a build error with your new code.
To make sure it's a clean build, I deleted completely and did a fresh checkout (even incl. the idf).

This is the error I get:
/root/temp/cspot/cspot/bell/main/io/BellHTTPServer.cpp: In member function 'virtual bool bell::BellHTTPServer::handleGet(CivetServer*, mg_connection*)':
/root/temp/cspot/cspot/bell/main/io/BellHTTPServer.cpp:151:28: error: exception handling disabled, use '-fexceptions' to enable
  151 |   } catch (std::exception& e) {
      |                            ^
In file included from /root/temp/cspot/cspot/bell/main/io/include/BellHTTPServer.h:3,
                 from /root/temp/cspot/cspot/bell/main/io/BellHTTPServer.cpp:1:
/root/temp/cspot/cspot/bell/main/io/BellHTTPServer.cpp:152:71: error: 'e' was not declared in this scope; did you mean 'std::numbers::e'?
  152 |     BELL_LOG(error, "HttpServer", "Exception occured in handler: %s", e.what());
      |                                                                       ^
/root/temp/cspot/cspot/bell/main/utilities/include/BellLogger.h:145:54: note: in definition of macro 'BELL_LOG'
  145 |     bell::bellGlobalLogger->type(__FILE__, __LINE__, __VA_ARGS__); \
      |                                                      ^~~~~~~~~~~
In file included from /root/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/bits/max_size_type.h:37,
                 from /root/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/bits/ranges_base.h:38,
                 from /root/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/string_view:50,
                 from /root/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/bits/basic_string.h:47,
                 from /root/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/string:53,
                 from /root/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/bits/locale_classes.h:40,
                 from /root/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/bits/ios_base.h:41,
                 from /root/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/ios:42,
                 from /root/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/istream:38,
                 from /root/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/sstream:38,
                 from /root/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/chrono:41,
                 from /root/temp/cspot/cspot/bell/main/utilities/include/BellLogger.h:6:
/root/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/numbers:122:27: note: 'std::numbers::e' declared here
  122 |   inline constexpr double e = e_v<double>;
      |                           ^
/root/temp/cspot/cspot/bell/main/io/BellHTTPServer.cpp: In member function 'virtual bool bell::BellHTTPServer::handlePost(CivetServer*, mg_connection*)':
/root/temp/cspot/cspot/bell/main/io/BellHTTPServer.cpp:184:71: error: 'e' was not declared in this scope; did you mean 'std::numbers::e'?
  184 |     BELL_LOG(error, "HttpServer", "Exception occured in handler: %s", e.what());
      |                                                                       ^
/root/temp/cspot/cspot/bell/main/utilities/include/BellLogger.h:145:54: note: in definition of macro 'BELL_LOG'
  145 |     bell::bellGlobalLogger->type(__FILE__, __LINE__, __VA_ARGS__); \
      |                                                      ^~~~~~~~~~~
/root/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/numbers:122:27: note: 'std::numbers::e' declared here
  122 |   inline constexpr double e = e_v<double>;
      |                           ^
[781/1308] Building C object esp-idf/c...f_console.dir/argtable3/arg_file.c.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /root/temp/cspot/targets/esp32/build/log/idf_py_stderr_output_13614 and /root/temp/cspot/targets/esp32/build/log/idf_py_stdout_output_13614

tobiasguyer, 2023-12-25:
Have you done the stanard setup?
Clone the repository with --recursive flag, move to targets/esp32.. idf.py set-target esp32.. idf.py menuconfig, set wifi and password and then build.. 
i think i had the same problems once, when i forgot to set the target to esp32. I think this is also needed for some bell::functions..
And just to get it right, are you also using espidf 4.4 with a esp32 wrover with 4mb psram? I'm not sure, if it would work with 8mb psram, could be changed in the sdkconfig otherwise..

xris99, 2023-12-26:
Thank you so much @tobiasguyer for your help. I had an issue with my build environment, namely the protobuf version installed.

If others run into issues as well, make sure you have the followling:
After 
```
idf.py add-dependency "espressif/mdns^1.2.2" 
```
in the instructions above, make sure you have protobuf installed in version 3.20.1
```
apt-get install protobuf-compiler
pip install python3-protobuf
pip install protobuf==3.20.1
 ```
then continue with the instructions above starting with
```
idf.py set-target esp32
```

Thank you so much for your help. Will try on the device now.

myreauks, 2024-01-04:
I finally had the chance to flash this on device. I have not yet connected to a DAC, but Spotify connects successfully to the ESP and starts playing so it seems very promising.

LeoSum8, 2024-01-09:
Thank you @tobiasguyer for the fork and @playduck for the detailed steps.
However I can't complete the build as I am running into errors when following the steps in https://github.com/feelfreelinux/cspot/issues/156#issuecomment-1866629464

I am not 100% sure what the final CMakeLists.txt should look like as I don't really understand the second change that you make. Currently I have:

```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

# Don't override EXTRA_COMPONENT_DIRS as platformio uses it.  Instead we append
# see https://github.com/platformio/platform-espressif32/issues/341
list(APPEND EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)
# list(APPEND EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/led_strip)

if(NOT IDF_NO_INCLUDE)
    include($ENV{IDF_PATH}/tools/cmake/project.cmake)
#    REQUIRES led_strip mdns spiffs esp_wifi nvs_flash protocol_examples_common
endif()
project(cspot-esp32)
```
As you can see, I commented out the line `REQUIRES led_strip mdns spiffs esp_wifi nvs_flash protocol_examples_common` since it leads to an error:

```
leosum8@macmini:~/cspot/cspot/targets/esp32$ idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/leosum8/cspot/cspot/targets/esp32/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig will be created.
Running cmake in directory /home/leosum8/cspot/cspot/targets/esp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/leosum8/.espressif/python_env/idf5.3_py3.11_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/leosum8/cspot/cspot/targets/esp32"...
-- Found Git: /usr/bin/git (found version "2.43.0") 
CMake Error at /home/leosum8/esp/esp-idf/tools/cmake/project.cmake:465 (message):
  Directory specified in EXTRA_COMPONENT_DIRS doesn't exist:
  /home/leosum8/esp/esp-idf/examples/common_components/led_strip
Call Stack (most recent call first):
  /home/leosum8/esp/esp-idf/tools/cmake/project.cmake:549 (__project_init)
  CMakeLists.txt:13 (project)

-- Configuring incomplete, errors occurred!
HINT: The component with path specified in the EXTRA_COMPONENT_DIRS variable has been moved to IDF component manager (or has been removed).
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details.
cmake failed with exit code 1, output of the command is in the /home/leosum8/cspot/cspot/targets/esp32/build/log/idf_py_stderr_output_7503 and /home/leosum8/cspot/cspot/targets/esp32/build/log/idf_py_stdout_output_7503
```

when commenting out and repeating the set-target, I get:
```
leo@macmini:~/cspot/cspot/targets/esp32$ idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Directory '/home/leo/cspot/cspot/targets/esp32/build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.
```
manually removing the folder doesn't get rid of this error, I reappers every time.

Despite this error I can continue with menuconfig and start building. But building fails with:
```
...
[1320/1672] Building CXX object esp-idf/main/cspot/bell/CMakeFiles/bell.dir/main/io/BellHTTPServer.cpp.obj
FAILED: esp-idf/main/cspot/bell/CMakeFiles/bell.dir/main/io/BellHTTPServer.cpp.obj 
/home/leosum8/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM -DFMT_HEADER_ONLY -DIDF_VER=\"v5.3-dev-1288-g5524b692ee\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DPB_ENABLE_MALLOC -DPB_FIELD_32BIT -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -DUSE_DEFAULT_STDLIB="1 -DTARGET_OS_IPHONE=0" -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/leosum8/cspot/cspot/targets/esp32/build/config -I/home/leosum8/cspot/cspot/cspot/bell/main/audio-codec/include -I/home/leosum8/cspot/cspot/cspot/bell/main/audio-dsp/include -I/home/leosum8/cspot/cspot/cspot/bell/main/audio-sinks/include -I/home/leosum8/cspot/cspot/cspot/bell/main/io/include -I/home/leosum8/cspot/cspot/cspot/bell/main/utilities/include -I/home/leosum8/cspot/cspot/cspot/bell/main/platform -I/home/leosum8/cspot/cspot/cspot/bell/external/mqtt/include -I/home/leosum8/cspot/cspot/cspot/bell/main/audio-containers/include -I/home/leosum8/cspot/cspot/cspot/bell/main/audio-sinks/include/esp -I/home/leosum8/cspot/cspot/cspot/bell/external/civetweb/include -I/home/leosum8/cspot/cspot/targets/esp32/build/esp-idf/main/cspot/bell -I/home/leosum8/esp/esp-idf/components/newlib/platform_include -I/home/leosum8/esp/esp-idf/components/freertos/config/include -I/home/leosum8/esp/esp-idf/components/freertos/config/include/freertos -I/home/leosum8/esp/esp-idf/components/freertos/config/xtensa/include -I/home/leosum8/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/leosum8/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/leosum8/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/home/leosum8/esp/esp-idf/compo [... cut ...]
  151 |   int alternativeCount, filesCount = 0;
      |       ^~~~~~~~~~~~~~~~
/home/leosum8/cspot/cspot/cspot/src/TrackQueue.cpp:152:8: warning: unused variable 'canPlay' [-Wunused-variable]
  152 |   bool canPlay = false;
      |        ^~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/leosum8/cspot/cspot/targets/esp32/build/log/idf_py_stderr_output_8386 and /home/leosum8/cspot/cspot/targets/esp32/build/log/idf_py_stdout_output_8386
```

Any ideas?

playduck, 2024-01-09:
Did you successfully download the dependencies?
```
idf.py add-dependency "espressif/led_strip^2.5.2"
idf.py add-dependency "espressif/mdns^1.2.2"
```

---

My CMake files look like this:

`targets/esp32/CMakeLists.txt`
```CMake
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

# Don't override EXTRA_COMPONENT_DIRS as platformio uses it.  Instead we append
# see https://github.com/platformio/platform-espressif32/issues/341
list(APPEND EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)
# list(APPEND EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/led_strip)

if(NOT IDF_NO_INCLUDE)
    include($ENV{IDF_PATH}/tools/cmake/project.cmake)
endif()
project(cspot-esp32)
```

`targets/esp32/main/CMakeLists.txt`
```CMake
cmake_minimum_required(VERSION 3.5)

# CMake options
set(CMAKE_CXX_STANDARD 17)
# Main target sources
file(GLOB SOURCES "*.cpp" "*.c")

# Configure the target
idf_component_register(
    SRCS ${SOURCES}
    INCLUDE_DIRS "."
    REQUIRES led_strip mdns spiffs esp_wifi nvs_flash protocol_examples_common
)
idf_build_set_property(COMPILE_OPTIONS "-fdiagnostics-color=always" APPEND)

# Build static library, do not build test executables
option(BUILD_SHARED_LIBS OFF)
option(BUILD_TESTING OFF)

# Import cspot library
add_subdirectory("../../../cspot" ${CMAKE_CURRENT_BINARY_DIR}/cspot)

# Configure the target
target_link_libraries(${COMPONENT_LIB} PUBLIC cspot)
target_compile_options(${COMPONENT_LIB} PRIVATE -std=gnu++17)
```

LeoSum8, 2024-01-09:
Hi, and thanks for answering!
The dependencies are installed. 

Thanks to your CMakeLists.txt files which I copied the pre-building errors during set-target command went away. 

However building still fails with the same error.

I also tried starting from scratch. 
Also following the steps above to install protobuf in the right version  (https://github.com/feelfreelinux/cspot/issues/156#issuecomment-1869431483) doesn't seem to change anything.

```shell

...

[libprotobuf WARNING google/protobuf/compiler/parser.cc:646] No syntax specified for the proto file: protobuf/login5.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
/home/leosum8/cspot/cspot/targets/esp32/build/esp-idf/main/cspot/nanopb/generator/nanopb_generator.py:22: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources # pyinstaller / protobuf 2.5 seem to need these
[730/1672] Running C++ protocol buffer compiler using nanopb plugin on /home/leosum8/cspot/cspot/cspot/protobuf/spirc.proto
[libprotobuf WARNING google/protobuf/compiler/parser.cc:646] No syntax specified for the proto file: protobuf/spirc.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
/home/leosum8/cspot/cspot/targets/esp32/build/esp-idf/main/cspot/nanopb/generator/nanopb_generator.py:22: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources # pyinstaller / protobuf 2.5 seem to need these
[1274/1672] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/EspPlayer.cpp.obj
/home/leosum8/cspot/cspot/targets/esp32/main/EspPlayer.cpp: In lambda function:
/home/leosum8/cspot/cspot/targets/esp32/main/EspPlayer.cpp:71:19: warning: unused variable 'volume' [-Wunused-variable]
   71 |               int volume = std::get<int>(event->data);
      |                   ^~~~~~
[1276/1672] Building CXX object esp-idf/main/cspot/CMakeFiles/cspot.dir/src/MercurySession.cpp.obj
/home/leosum8/cspot/cspot/cspot/src/MercurySession.cpp: In member function 'cspot::MercurySession::Response cspot::MercurySession::decodeResponse(const std::vector<unsigned char>&)':
/home/leosum8/cspot/cspot/cspot/src/MercurySession.cpp:222:8: warning: unused variable 'sequenceLength' [-Wunused-variable]
  222 |   auto sequenceLength = ntohs(extract<uint16_t>(data, 0));
      |        ^~~~~~~~~~~~~~
/home/leosum8/cspot/cspot/cspot/src/MercurySession.cpp:225:8: warning: unused variable 'partsNumber' [-Wunused-variable]
  225 |   auto partsNumber = ntohs(extract<uint16_t>(data, 11));
      |        ^~~~~~~~~~~
[1277/1672] Building CXX object esp-idf/main/cspot/CMakeFiles/cspot.dir/src/AccessKeyFetcher.cpp.obj
/home/leosum8/cspot/cspot/cspot/src/AccessKeyFetcher.cpp: In member function 'void cspot::AccessKeyFetcher::updateAccessKey()':
/home/leosum8/cspot/cspot/cspot/src/AccessKeyFetcher.cpp:123:13: warning: unused variable 'expiresIn' [-Wunused-variable]
  123 |         int expiresIn = loginResponse.response.ok.access_token_expires_in / 2;
      |             ^~~~~~~~~
[1280/1672] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj
In file included from /home/leosum8/esp/esp-idf/components/driver/deprecated/driver/i2s.h:23,
                 from /home/leosum8/cspot/cspot/cspot/bell/main/audio-sinks/include/esp/adac.h:12,
                 from /home/leosum8/cspot/cspot/cspot/bell/main/audio-sinks/include/esp/AC101AudioSink.h:12,
                 from /home/leosum8/cspot/cspot/targets/esp32/main/main.cpp:44:
/home/leosum8/esp/esp-idf/components/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
/home/leosum8/esp/esp-idf/components/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
[1294/1672] Building CXX object esp-idf/main/cspot/CMakeFiles/cspot.dir/src/TrackPlayer.cpp.obj
/home/leosum8/cspot/cspot/cspot/src/TrackPlayer.cpp: In member function 'virtual void cspot::TrackPlayer::runTask()':
/home/leosum8/cspot/cspot/cspot/src/TrackPlayer.cpp:192:15: warning: unused variable 'r' [-Wunused-variable]
  192 |       int32_t r =
      |               ^
[1301/1672] Building CXX object esp-idf/main/cspot/bell/CMakeFiles/bell.dir/main/io/BellMQTTClient.cpp.obj
/home/leosum8/cspot/cspot/cspot/bell/main/io/BellMQTTClient.cpp: In member function 'void bell::MQTTClient::connect(const std::string&, uint16_t, const std::string&, const std::string&)':
/home/leosum8/cspot/cspot/cspot/bell/main/io/BellMQTTClient.cpp:36:7: warning: unused variable 'status' [-Wunused-variable]
   36 |   int status = fcntl(socket.getFd(), F_SETFL,
[... cut ...]
  151 |   int alternativeCount, filesCount = 0;
      |       ^~~~~~~~~~~~~~~~
/home/leosum8/cspot/cspot/cspot/src/TrackQueue.cpp:152:8: warning: unused variable 'canPlay' [-Wunused-variable]
  152 |   bool canPlay = false;
      |        ^~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/leosum8/cspot/cspot/targets/esp32/build/log/idf_py_stderr_output_40092 and /home/leosum8/cspot/cspot/targets/esp32/build/log/idf_py_stdout_output_40092

```

tobiasguyer, 2024-01-10:
If you have  installed all the dependencies, and have done a full-clean too, there's the possibility, that you'll have to delete the existing build folder in targets/esp32.  Otherwise, there should be a CMakeLog- and CMakeError file in the Build folder, that could help further.
And i'm sorry about the led-strip thing. I haven't implemented anything status-led-like, so there's no need for that dependency. My CMakeFile hasn't any REQUIRES. Could be, because i'm still using esp-idf 4.4

myreauks, 2024-02-06:
I verified this working with sound output, but all of a sudden Spotify cannot connect to the ESP32 anymore. Here is the output from the monitor, Spotify just gets stuck to connecting. Tried with desktop app and iOS app.

```
[1970-01-01 00:00:05.390] I BellHTTPServer.cpp:191: Server listening on port 7864
Waiting for spotify app to connect...
[1970-01-01 00:00:11.876] I main.cpp:117: Received zeroauth POST data
[1970-01-01 00:00:12.422] D Session.cpp:67: Connecting with AP <ap-gew4.spotify.com:4070>
[1970-01-01 00:00:12.471] D PlainConnection.cpp:101: Connected to spotify server
[1970-01-01 00:00:12.543] I Session.cpp:43: Received APHello response
[1970-01-01 00:00:12.579] D Session.cpp:48: Received shannon keys
[1970-01-01 00:00:12.669] D Session.cpp:87: Authorization successful
[1970-01-01 00:00:12.684] I MercurySession.cpp:42: Received packet, command: 4
[1970-01-01 00:00:12.691] D TimeProvider.cpp:15: Time synced with spotify servers
[1970-01-01 00:00:12.701] I MercurySession.cpp:42: Received packet, command: 2
[1970-01-01 [00:00:121970-.70701] -D 01 MercurySession.cpp00:251: Executing Mercury Request, type SUB
:00:12.708] I MercurySession.cpp:42: Received packet, command: 118
[1970-01-01 00:00:12.727] I MercurySession.cpp:42: Received packet, command: 27
[1970-01-01 00:00:12.739] D MercurySession.cpp:153: Received country code FI
[1970-01-01 00:00:12.781] I MercurySession.cpp:42: Received packet, command: 80
[1970-01-01 00:00:12.783] I MercurySession.cpp:42: Received packet, command: 31
[1970-01-01 00:00:12.790] I MercurySession.cpp:42: Received packet, command: 105
[1970-01-01 00:00:12.800] I AccessKeyFetcher.cpp:99: Access token expired, fetching new one... 288
[1970-01-01 00:00:12.810] I MercurySession.cpp:42: Received packet, command: 181
[1970-01-01 00:00:12.823] I MercurySession.cpp:42: Received packet, command: 74
[1970-01-01 00:00:12.835] I MercurySession.cpp:42: Received packet, command: 179
[1970-01-01 00:00:12.839] D MercurySession.cpp:174: Received mercury packet
[1970-01-01 00:00:12.849] D MercurySession.cpp:251: Executing Mercury Request, type SEND
[1970-01-01 00:00:12.871] D SpircHandler.cpp:61: Sent kMessageTypeHello!
[1970-01-01 00:00:12.994] I MercurySession.cpp:42: Received packet, command: 178
[1970-01-01 00:00:12.996] D MercurySession.cpp:174: Received mercury packet
[1970-01-01 00:00:13.363] I AccessKeyFetcher.cpp:114: Access token sucessfully fetched
```

myreauks, 2024-02-07:
And now it is connecting successfully again. Very mysterious.

myreauks, 2024-03-17:
An update on the build after I've been using it for a while. It seems to be running a bit unstable at the moment. If I pause a track and then play another, most of the time the player stops working a this point and every request after results in the following line in the monitor:
`[1970-01-01 00:01:25.916] I MercurySession.cpp:42: Received packet, command: 181`

Here is the whole output before getting unresponsive. Any ideas what the cause might be?
```
I (11070) cspot: Connected to AP, start spotify receiver
[1970-01-01 00:00:09.845] I BellHTTPServer.cpp:191: Server listening on port 7864
Waiting for spotify app to connect...
[1970-01-01 00:00:32.810] I main.cpp:133: Received zeroauth POST data
[1970-01-01 00:00:33.445] D Session.cpp:67: Connecting with AP <ap-gew4.spotify.com:4070>
[1970-01-01 00:00:33.494] D PlainConnection.cpp:101: Connected to spotify server
[1970-01-01 00:00:33.565] I Session.cpp:43: Received APHello response
[1970-01-01 00:00:33.602] D Session.cpp:48: Received shannon keys
[1970-01-01 00:00:33.711] D Session.cpp:87: Authorization successful
[1970-01-01 00:00:33.731] I MercurySession.cpp:42: Received packet, command: 4
[1970-01-01 00:00:33.739] D TimeProvider.cpp:15: Time synced with spotify servers
[1970-01-01 00:00:33.748] I MercurySession.cpp:42: Received packet, command: 2
[1970-01-01 [00:00:331970.-755] 01-D 01 00:MercurySession.cpp00:251: :Executing Mercury Request, type SUB33
.755] I MercurySession.cpp:42: Received packet, command: 118
[1970-01-01 00:00:33.774] I MercurySession.cpp:42: Received packet, command: 27
[1970-01-01 00:00:33.783] D MercurySession.cpp:153: Received country code FI
[1970-01-01 00:00:33.829] I MercurySession.cpp:42: Received packet, command: 80
[1970-01-01 00:00:33.832] I MercurySession.cpp:42: Received packet, command: 31
[1970-01-01 00:00:33.838] I MercurySession.cpp:42: Received packet, command: 105
[1970-01-01 00:00:33.850] I MercurySession.cpp:42: Received packet, command: 181
[1970-01-01 00:00:33.857] I MercurySession.cpp:42: Received packet, command: 74
[1970-01-01 00:00:33.866] I MercurySession.cpp:42: Received packet, command: 179
[[1970-197001--01 01-00:01 00:00:33.00:875] 33I .875AccessKeyFetcher.cpp] :99: Access token expired, fetching new one... 288D 
MercurySession.cpp:174: Received mercury packet
[1970-01-01 00:00:33.897] D MercurySession.cpp:251: Executing Mercury Request, type SEND
[1970-01-01 00:00:33.919] D SpircHandler.cpp:61: Sent kMessageTypeHello!
[1970-01-01 00:00:34.022] I MercurySession.cpp:42: Received packet, command: 181
[1970-01-01 00:00:34.025] D SpircHandler.cpp:69: Received subscription response
[1970-01-01 00:00:34.033] D SpircHandler.cpp:132: Notify frame
[1970-01-01 00:00:34.062] I MercurySession.cpp:42: Received packet, command: 178
[1970-01-01 00:00:34.063] D MercurySession.cpp:174: Received mercury packet
[1970-01-01 00:00:34.292] I MercurySession.cpp:42: Received packet, command: 181
[1970-01-01 00:00:34.298] D SpircHandler.cpp:69: Received subscription response
[1970-01-01 00:00:34.313] D SpircHandler.cpp:179: Load frame 10!
[1970-01-01 00:00:34.324] D MercurySession.cpp:251: Executing Mercury Request, type SEND
[1970-01-01 00:00:34.343] I TrackPlayer.cpp:98: Resetting state
[1970-01-01 00:00:34.432] I AccessKeyFetcher.cpp:114: Access token sucessfully fetched
[1970-01-01 00:00:34.439] D MercurySession.cpp:251: Executing Mercury Request, type GET
[1970-01-01 00:00:34.481] I MercurySession.cpp:42: Received packet, command: 178
[1970-01-01 00:00:34.483] D MercurySession.cpp:174: Received mercury packet
[1970-01-01 00:00:34.765] I MercurySession.cpp:42: Received packet, command: 178
[1970-01-01 00:00:34.767] D MercurySession.cpp:174: Received mercury packet
[1970-01-01 00:00:34.774] I TrackQueue.cpp:158: Track name: Reckoner
[1970-01-01 00:00:34.781] I TrackQueue.cpp:159: Track duration: 290213
[1970-01-01 00:00:34.789] D TrackQueue.cpp:161: trackInfo.restriction.size() = 1
[1970-01-01 00:00:34.799] D TrackQueue.cpp:212: File format: 2
[1970-01-01 00:00:34.806] D TrackQueue.cpp:212: File format: 1
[1970-01-01 00:00:34.860] I MercurySession.cpp:42: Received packet, command: 13
[1970-01-01 00:00:34.862] I TrackQueue.cpp:252: Got audio key
[1970-01-01 00:00:34.868] I TrackQueue.cpp:275: Received access key, fetching CDN URL...
[1970-01-01 00:00:35.355] I TrackQueue.cpp:301: Received CDN URL, https://audio-ak-spotify-com.akamaized.net/audio/67084030365ab0cf7b14b9e54998835b3249dc90?__token__=exp=1710763279~hmac=8edcf44032d3f294061a647999d20469bc296f4cd0dc71a8e987c3ce4d3bd566
[1970-01-01 00:00:35.370] I TrackPlayer.cpp:171: Got track ID=67084030365ab0cf7b14b9e54998835b3249dc90
[1970-01-01 00:00:35.382] I CDNAudioFile.cpp:43: Opening HTTP stream to https://audio-ak-spotify-com.akamaized.net/audio/67084030365ab0cf7b14b9e54998835b3249dc90?__token__=exp=1710763279~hmac=8edcf44032d3f294061a647999d20469bc296f4cd0dc71a8e987c3ce4d3bd566
[1970-01-01 00:00:35.473] D MercurySession.cpp:251: Executing Mercury Request, type GET
[1970-01-01 00:00:35.582] I MercurySession.cpp:42: Received packet, command: 178
[1970-01-01 00:00:35.584] D MercurySession.cpp:174: Received mercury packet
[1970-01-01 00:00:35.592] I TrackQueue.cpp:158: Track name: House Of Cards
[1970-01-01 00:00:35.599] I TrackQueue.cpp:159: Track duration: 328293
[1970-01-01 00:00:35.607] D TrackQueue.cpp:161: trackInfo.restriction.size() = 1
[1970-01-01 00:00:35.617] D TrackQueue.cpp:212: File format: 2
[... cut ...]
[1970-01-01 00:00:47.271] D MercurySession.cpp:251: Executing Mercury Request, type GET
[1970-01-01 00:00:47.317] I MercurySession.cpp:42: Received packet, command: 178
[1970-01-01 00:00:47.349] I MercurySession.cpp:42: Received packet, command: 178
[1970-01-01 00:00:47.420] I MercurySession.cpp:42: Received packet, command: 181
[1970-01-01 00:00:57.216] I MercurySession.cpp:42: Received packet, command: 181
[1970-01-01 00:00:58.882] I MercurySession.cpp:42: Received packet, command: 181
[1970-01-01 00:01:25.916] I MercurySession.cpp:42: Received packet, command: 181
```

tobiasguyer, 2024-07-10:
Sorry for not responding for so long.
Too be honest, i don't work with a pcm audio sink, thus my main structure changed a lot from the main branch, but i should have written a pcm adaption of my sink once. I'll try looking into it, but their are still some commits to do, before i can focus on your problem.
Until then, tinkering with the tasksize (main cpp:144 and EspPlayer.cpp:21) and the buffersize(EspPlayer.cpp:28) could hav a huge impact

tobiasguyer, 2024-07-11:
So.. I haven't tested it properly, I just made a serial output of the pcm-stream.
But it seems to me, the main problem was in the mercury manager with gzipping. Their were truncated messages, that weren't decoded properly. That should be solved with my newest commit. Let me know if there are still problems

myreauks, 2024-07-28:
Thanks! I'll try the newest commit in the coming days and report back.

myreauks, 2024-07-31:
It seems that the player is not getting info on further tracks, it just stops playing after three tracks. Here is the output at point of stopping:
```
.[0;34mI .[0;90mCDNAudioFile.cpp.[0m:70: .[0mHeader and footer bytes received
.[0;34mI .[0;90mTrackPlayer.cpp.[0m:267: .[0mPlaying
.[0;31mD .[0;37mMercurySession.cpp.[0m:301: Executing Mercury Request, type SEND
.[0;34mI .[0;37mMercurySession.cpp.[0m:42: .[0mReceived packet, command: 178
.[0;31mD .[0;37mMercurySession.cpp.[0m:175: Received mercury packet
.[0;31mD .[0;37mMercurySession.cpp.[0m:260: Creating new Mercury Response, seq: 76, flags: 1, parts: 1
.[0;34mI .[0;37mMercurySession.cpp.[0m:42: .[0mReceived packet, command: 4
.[0;31mD .[0;91mTimeProvider.cpp.[0m:15: Time synced with spotify servers
.[0;34mI .[0;37mMercurySession.cpp.[0m:42: .[0mReceived packet, command: 74
.[0;34mI .[0;37mMercurySession.cpp.[0m:42: .[0mReceived packet, command: 4
.[0;31mD .[0;91mTimeProvider.cpp.[0m:15: Time synced with spotify servers
.[0;34mI .[0;37mMercurySession.cpp.[0m:42: .[0mReceived packet, command: 74
.[0;34mI .[0;90mTrackPlayer.cpp.[0m:304: .[0mEOF
.[0;34mI .[0;90mTrackPlayer.cpp.[0m:346: .[0mPlaying done
```

tobiasguyer, 2024-08-01:
I'm so sorry, i was redoing a lot of the cspot structure and made some mistakes. my current build(master or EventManager-branch) should run stable and with no stop for quite some time. it's still messy, but currently @feelfreelinux is looking over the build.
i think, it's gonna take a while, until it's properly reviewed and merged in the main branch. i hope, until then, my build is gonna suffice for you.
in my build, their should come no changes for quite some time now, at least, if you guys experience stable runs too.
have fun and i hope it works properly

myreauks, 2024-08-01:
Thanks for the update, for some reason it is still playing only three tracks at a time. However, it seems to be very stable, I got just one random crash which I think has something to do with the task size or the buffer size. I'll try playing with those values. 
Spotify UI is still not updating the now playing or the queue for whatever reason. At the moment that's not a big deal though, as it doesn't affect the playback. 

Thanks a lot for your efforts!

tobiasguyer, 2024-09-29:
I'm sorry, i hope you haven't tried too much already. I had missunderstood the preloadedTracks structure, and misswrote some lines.
Now, at least the cliBuild, and the VS1053 are working.
I've tried to adjust the espPlayer.cpp as good as possible, but i have no ways to test it.
There is still a issue or two, but it works quite amazing and has plenty of new futures.
Let me know how it works. 
And if it doesn't, and i won't respond fast enough, take a look at the cliPlayer.cpp

tobiasguyer, 2024-11-06:
So.. i've ported my build to espidf v5.3. With 5.1 it had a build error, so be carefull to use the latest espidf. Because of some issues with mbedtls, i had to change quite a lot in cspot/bell. So be sure, to update the submodules after porting to the newest commit.
https://github.com/tobiasguyer/cspot
have fun and let me now, if it is finally working on esp. it works, apart from the classic shuffeling (no smart shuffle) on cli(WIN32) and esp32 with vs1053