feelfreelinux/cspot #163: ImportError: attempted relative import with no known parent package

anim1311, 2023-10-27:
I am trying to build and flash my esp32 with esp-idf V4.4 and I get the following error

anim1311, 2023-10-27:
[image]
Here is the error

philippe44, 2023-10-30:
How did you install protobuf?

anim1311, 2023-10-31:
With pip install protobuf

philippe44, 2023-10-31:
It's a Windows machine, correct?

MMiziniak, 2024-01-20:
Did You solv problem?
I have same error.
protobuf with pip install protobuf

alyssadev, 2024-06-17:
I encountered this issue on linux with esp-idf, it was resolved by installing grpcio-tools into esp-idf's venv (which is set up automatically with `. ./export.sh`):

```
$ which pip3
/home/aly/.espressif/python_env/idf5.4_py3.12_env/bin/pip3
$ pip3 install protobuf grpcio.tools
```